jbatonnet/Rinkhals #138: Problems with timelapse

Mikmikbt, 2025-04-16:
Hi, I found a lack in firmware and Mainsail.
I cannot activate the USB recording of the timelapse in any way.
Even if I load gcode on USB and try to start printing from the display.
The error states that it is not possible to start the timelapse because the webcam is already busy (with Mainsail).
Can you please solve this?
If I could run the timelapse even with Mainsail, you could almost do without the Slicer Next and the original app and use only Orca and Mainsail or Octoeverywhere.
Thank you very much

jbatonnet, 2025-04-16:
Some people have been talking about timelapse on Discord, but I have no idea how it works in Mainsail and other. I'd need people to look at that in details to see what's needed, if it's possible.
For the busy webcam error, check the FAQ for camera not available in Anycubic apps, same answer.
Closing this one for now as thing work as expected

closed 2025-04-16 by jbatonnet

derveloper, 2025-04-18:
I (chatgpt) workarounded the problem with this handy ffmpeg oneliner

```
ffmpeg -fflags nobuffer -seekable 1 -i "http://192.168.10.140/webcam1/?action=stream" -i "http://192.168.10.140/webcam0/?action=stream" -filter_complex " [0:v]select='not(mod(n\,10))',setpts=N/FRAME_RATE/TB[top]; [1:v]select='not(mod(n\,10))',setpts=N/FRAME_RATE/TB[bot]; [top][bot]vstack=inputs=2[v]" -map "[v]" -vsync vfr -c:v libx264 -preset veryfast -crf 23 -pix_fmt yuv420p -hide_banner -loglevel error -f matroska -live 1 timelapse_duo.mkv
```

Mikmikbt, 2025-04-19:
Forgive me, but where should this code be inserted? Does it serve to have
the opportunity to view the webcam from both Anycubic and Mainsail apps?

derveloper, 2025-04-19:
@Mikmikbt in case you on mac or linux, chances are hight that you can simply run this from a terminal. you might need to change the ip address to you local setup. on the printer itself you also have ffmpeg and you can run it there via ssh.

Mikmikbt, 2025-04-19:
Hello, I have windows and I use Mainsail. How do I proceed?

derveloper, 2025-04-19:
@Mikmikbt This might be a good opportunity to learn how to run terminal commands on Windows — take a look at WSL2 (Windows Subsystem for Linux).
That’s about as far as I can help you here.