jbatonnet/Rinkhals #141: KS1 - after a few reboots, start experiencing issues with Moonraker and then Rinkhals disappears

duncanch89, 2025-04-20:
**Describe the bug**
A clear and concise description of what the bug is, and what you expected to happen.

KS1 - 2.5.1.6

On my 4th install of Rinkhals, it keeps disappearing from the UI after I've rebooted my printer a few times.

Before Rinkhals disappears entirely, I will start experiencing error messages in Mainsail and Fluidd - "cannot connect to  MCU", it suggests a firmware restart and a device reboot, but the only thing that has fixed this issue once I start encountering it is doing a system restore and reinstalling Rinkhals, or leaving the printer powered off overnight.

**To Reproduce**
Steps to reproduce the behavior:

1. Install Rinkhals
2. Do some print jobs
3. Reboot printer
4. Get error message, try rebooting the firmware and the device a few times to see if it goes away
5. Rinkhals gone

**Logs / Additional information**

[debug-bundle_3467947133_19700101-080443.zip](https://github.com/user-attachments/files/19826322/debug-bundle_3467947133_19700101-080443.zip)

Thanks for everything you're doing!

stylon, 2025-04-21:
That sounds similar to what I experienced on my KS1 with Rinkhals. I decided to reinstall and Rinkhals appeared back again. See also issue #92.

duncanch89, 2025-04-21:
> That sounds similar to what I experienced on my KS1 with Rinkhals. I decided to reinstall and Rinkhals appeared back again. See also issue [#92](https://github.com/jbatonnet/Rinkhals/issues/92).

Agree, looks similar to my issues, I also seem to have lots of issues with wifi connectivity like you describe.

I will have a situation where the printer is powered on, shows as connected to network on the touch panel, and see it as on online device in my router devices screen, but Anycubic next slicer declaring it "offline" in both LAN and cloud versions (I use lan-to-cloud bridge, so when everything works properly, it shows up as "online" in both), and the various webUIs that come with Rinkhals won't work, but sometimes I am still able to connect to SSH into the machine despite all those other things not working. Very strange!

I have Unifi dream machine pro with Wifi 6 APs on a mesh network, both 2.4 and 5ghz modes enabled on the same SSID. By chance are you running a similar setup?

stylon, 2025-04-21:
FYI, I fixed my WIFI issues by replacing the "tiny" Anycubic antenna by a better one. It is still a PCB antenna, but one that has a length of roughly 9cm. Now, WIFI works like with all my other devices :-).

jbatonnet, 2025-04-24:
It could be related to MCU timing issues some people have, especially if you're running multiple Moonraker clients (Mainsail, Fluidd, OctoApp, ...)
Moonraker is really bad in terms of performance and scales even worse with multiple connections

duncanch89, 2025-04-24:
> Yes, indeed. I had similar WIFI issues. It behaved very sproadic for me, especially with my regular WIFI. What worked best for me was a dedicated, unencrypted 2.4GHz only router located directly next to the printer. I would have preferred a 100MB ethernet connection in that case.
> 
> FYI, I finally fixed my WIFI issues by replacing the "tiny" (foil-based) antenna by Anycubic by a better one. It is still a PCB antenna, but one that has a length of roughly 9cm and with a more sophisticated wire pattern. The same PCB worked well on an SBC before. Now, WIFI works like with all my other devices and even through 2 big walls with lots of metal :-).

Thanks for the tips. I'm continuing to have wifi connectivity issues even with the latest firmware, and after putting the printer on its own SSID with only 2.4ghz frequency.

I am going to replace my wifi antenna too and see if that helps. Does your replacement antenna resemble any of these?

[image]

duncanch89, 2025-04-24:
> Another idea, it could be related to MCU timing issues some people have, especially if you're running multiple Moonraker clients (Mainsail, Fluidd, OctoApp, ...) Moonraker is really bad in terms of performance and scales even worse with multiple connections

Good to know, thanks. I'd only ever used Octoprint prior to getting this printer, so I've been testing out multiple options until I settle on one. Going forward, I'll try only keeping one enabled at a time to see if that helps.

duncanch89, 2025-04-27:
This has happened again on the 2.5.2.3 firmware with latest Rinkhals re-installed.

I typically have to do a system restore to get the printer working properly again, however I found an old reddit thread of someone else having the MCU disconnect issue, and oddly enough one of the commenters said that what fixed the problem for them was replacing the printer's power cord.

When I set up my Kobra S1, I didn't bother swapping out the power cord from the old printer it was replacing - since it was already plugged into the wall, I left it there and plugged the existing cord into the back of the KS1. So, I went and found the power cord that came with the KS1 and swapped it in, and the MCU disconnect errors went away....

Still too early to tell, might have just gotten lucky - however at this point, I'm pretty sure this issue isn't related to Rinkhals so I'll close out the issue.

Thanks!

closed 2025-04-27 by duncanch89