jbatonnet/Rinkhals #142: Skew correction

Axos11, 2025-04-21:
Hello, first of all thanks for the work and effort put into thatrepo!

I wanted to ask if there is a possibility to add skew correction somehow? I tried tinkering arround with printer.cfg but it seems that it won't really accept any changes. (Kobra 3 on latest rinkhals and FW)

Thanks in advance!

Yangyuanxin, 2025-04-24:
This issue can only be supported by gklib. May I ask if the purpose of this feature is to improve device defects?

Axos11, 2025-05-06:
Hello @Yangyuanxin,

I always test my printer / filaments with https://vector3d.shop/products/califlower-calibration-tool-mk2?srsltid=AfmBOoqlxC39I0rM_T6KKMLPEzCqPjSmrTRMtTHB_-s2chGglMwZySH9 and there seems to be some skew that could be corrected. but it seems that the comandline won't recognize the commands for it.

jbatonnet, 2025-05-06:
Hi, more detailed instructions for printer configuration can be found here, even if I don't recommend tweaking these: https://www.notion.so/thedju/Printer-configuration-1e145e4b6fd880bfaee4c4d6e42421b2

I'm closing this issue as it's about configuration change and I won't provide support for it. Feel free to join the Discord server, many people are discussing this kind of changes!

closed 2025-05-06 by jbatonnet