jbatonnet/Rinkhals #143: Error 11407 after (I guess Vanilla Klipper Install)

taquion1264, 2025-04-22:
Hi,  I´ve tried to Install Rhinkals  and the first time  worked like a Charm , but when I installed all apps (even vanilla Klipper), started to fail  with the 11407, I´ve trying downgrading , eliminating the Rhinkhlas folder, etc, etc, without success , if you want some log I can provide it, I would like to install  Rhinklas cause I come from Klipper and I liked Minsail, and also, test some macros  if I can.

Would you help me with this ??, I also tried recovering the configs  but it didn´t worked

jbatonnet, 2025-04-22:
Hi, error 11407 is (almost?) always related to printer configuration being at fault.
I suggest you have a look at this Wiki page with many debugging options: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Printer-configuration#i-got-11407-or-my-printer-doesnt-boot-anymore

closed 2025-04-22 by jbatonnet

avxmw, 2025-06-01:
I've got the same issue, updated firmware to latest 2.4.04, rebooted clean, installed latest rinkhals release 20250531_01, rebooted clean, grabbed the latest vanilla klipper, rebooted and got prompted with error 11407.

Thankfully SSH was up and removing the ".enabled" and rebooting made the machine boot normally again. Putting the ".enabled" back produced the same error on next reboot.

I find the linked page a bit confusing, so this is what I've currently got on my machine, please tell me what to remove and try again, thanks.

```
root@Rockchip:~# find / -iname "printer*.cfg"
/tmp/printer.2.cfg
/tmp/printer.1.cfg
/userdata/app/gk/printer_mutable_20250426110309.cfg
/userdata/app/gk/printer_mutable.cfg
/userdata/app/gk/printer_data/config/default/printer.rinkhals.cfg
/userdata/app/gk/printer_data/config/default/printer.custom.cfg
/userdata/app/gk/printer_data/config/printer.custom.cfg
/userdata/app/gk/printer_data/config/printer.generated.cfg
/userdata/app/gk/printer.cfg
/userdata/app/gk/printer_mutable_20250405101919.cfg
/userdata/app/gk/printer_mutable_20250405101630.cfg
/userdata/app/gk/restore/printer_rebuild.cfg
/useremain/home/rinkhals/apps/vanilla-klipper/printer.klipper_K2P.cfg
/useremain/home/rinkhals/printer_data/config/printer.custom.cfg
/useremain/home/rinkhals/printer_data/config/printer.generated.cfg
/useremain/rinkhals/20250325_02/home/rinkhals/printer_data/config/printer.rinkhals.cfg
/useremain/rinkhals/20250325_02/home/rinkhals/printer_data/config/printer.custom.cfg
/useremain/rinkhals/20250424_01/home/rinkhals/printer_data/config/printer.rinkhals.cfg
/useremain/rinkhals/20250424_01/home/rinkhals/printer_data/config/printer.custom.cfg
/useremain/rinkhals/20250531_01-2/home/rinkhals/printer_data/config/printer.rinkhals.cfg
/useremain/rinkhals/20250531_01-2/home/rinkhals/printer_data/config/printer.custom.cfg
/useremain/rinkhals/20250531_01/home/rinkhals/printer_data/config/printer.rinkhals.cfg
/useremain/rinkhals/20250531_01/home/rinkhals/printer_data/config/printer.custom.cfg
/useremain/update_swu/vanilla-klipper/printer.klipper_K2P.cfg
```