jbatonnet/Rinkhals #146: Update for ACE Firmware on AnyCubic Kobra-S1

Error-Gap, 2025-04-23:
Allo Julien,

I am looking over the documentation regarding updating firmware on Kobra devices, including the S1.

Re: https://github.com/jbatonnet/Rinkhals/wiki/Rinkhals-%E2%80%90-Installation-and-firmware-updates

The documentation does not specify if the methods listed will work for ACE Pro device (multi-filament feeder) attached to the S1. As such, I am currently hesitant on using it for the ACE as I don't want to accidentally replace/break the firmware on the Kobra-S1 itself.

Can you confirm whether the method listed should also work for ACE updates?

Also, FYI it appears your personal website is down at https://www.thedju.net/
(sorry, I cannot find an email to notify you of this seperately)

jbatonnet, 2025-04-24:
Hi, I think it was a transient issue on my website, it's up again.

For the ACE Pro, I have actually never tries to do the update with this method. I always used the Anycubic OTA and it worked fine.
I usually leave my printer in LAN mode and when I validated the update for Rinkhals I open Cloud mode to get the updates and put LAN mode back.

I would suggest you join the Discord server. We have more than 700 members, there are good chances people tried that before and you'll get a clearer answer!

basvd, 2025-04-28:
I have performed all my AC firmware updates via the USB method and my ACE Pro firmware is still 1.3.76, which means it hasn't updated. I have tested updating with the ACE on (also off, but that obviously didn't make a difference).

My guess is the ACE update uses a different channel (perhaps OTA only). Maybe the printer also does a separate MQTT update check for it.

jbatonnet, 2025-04-28:
Some documentation here:
- https://github.com/utkabobr/DuckPro-Kobra3/issues/27
- https://github.com/printers-for-people/ACEResearch/issues/4

basvd, 2025-05-03:
To address the original question: The update process for the ACE should work the same way as the main firmware update process, given that you have a SWU package for the ACE. There are two versions available on: https://rinkhals.meowcat285.com/

It would be nice if we could archive all ACE firmware versions, so I've been trying to figure out the MQTT update check request.

closed 2025-05-07 by jbatonnet