jbatonnet/Rinkhals #148: After disabling mjpg-streamer Camera will not work after reboot

jmmerida, 2025-04-24:
As the title says. When I disable the app, to use the anycubic one at 720p it will work for the session. If I reboot the printer it will not detect the camera until I enable and disable again the app.

I am using the kobra 3 version 2.3.9.3

jbatonnet, 2025-04-24:
Okay I can repro now. It seems to be a startup timing issue.
I'll play a little bit with that to see if I can fix it

jbatonnet, 2025-04-24:
Okay not sure why but adding a 2s delay on startup fixes the issue.
On 5-6 reboots I now get 100% working, but it's not guaranteed. It might happen again, in this case just reboot the printer or enable/disable the mjpg app.

I'll push that in next release.

jbatonnet, 2025-04-24:
Available in last release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250424_01

closed 2025-04-24 by jbatonnet

jmmerida, 2025-04-24:
Thanks! I can confirm it is solved for me too!