jbatonnet/Rinkhals #150: S1C print hangs and cannot cancel print from webui or printer screen

Ejmathewp, 2025-04-25:
**Describe the bug**
Using the latest printer FW and Rinkhals, my print gradually slows until the toolhead is no longer moving at all. The temps hold at printing temps and the print cannot be cancelled through Fluidd/Mainsail or the printer screen. The only recovery is a power cycle. If I choose "continue print" after the power cycle, it again slows and stops shortly into the print.

**To Reproduce**
Steps to reproduce the behavior:
1. Start print
2. Wait
3. Printer stops moving

**Logs / Additional information**
Debug Bundle is attached.

[debug-bundle_1318143742_20250425-183051.zip](https://github.com/user-attachments/files/19914429/debug-bundle_1318143742_20250425-183051.zip)

Ejmathewp, 2025-04-26:
Downgraded to 2.4.8.3 and the compatible Rinkhals and the same file prints fine.

jbatonnet, 2025-05-07:
I cannot see anything weird in the logs. Usually when this happens it's a MCU desync caused by CPU / Memory being too busy on other things.
Let me know if it happens again

closed 2025-05-26 by jbatonnet