jbatonnet/Rinkhals #154: Adaptive mesh and level only object

entuccio, 2025-04-29:
Hi,

I’m using Rinkhals on an Anycubic Kobra S1 Combo.
I would like to confirm if the BED_MESH_CALIBRATE ADAPTIVE=1 command is fully supported.

The goal is to level bed mesh only the printed area to save time and improve first layer quality.

I have properly configured [bed_mesh] (mesh_min, mesh_max, adaptive_margin, algorithm: lagrange), and added BED_MESH_CALIBRATE ADAPTIVE=1 in the start G-code. but it doesn't work.

However, when triggering it before a print, the printer heats up but doesn’t start probing.

Could you please confirm:

Is ADAPTIVE=1 currently supported?

If not, are there plans to implement it?

Is there an official workaround for now?

Thanks a lot for your support and for the great work on Rinkhals!

swefd, 2025-04-29:
I'm also interested in this feature for Kobra 3

jbatonnet, 2025-05-12:
It doesn't seems to be implementer but it is on the Anycubic roadmap. I'll close this issue in the meantime!

closed 2025-05-12 by jbatonnet