jbatonnet/Rinkhals #156: New Firmware for K3 =  V2.4.0_1.3.9_1.3.9_release

FernandoMassa, 2025-04-30:
I had the correct firmware and was able to root it perfectly, but then I went to the "printer information" screen and a message appeared, asking me to update the firmware. Like a fool I clicked ok and after that, it overwrote everything and now I'm afraid to do the process again because the firmware is different.

Does anyone know if this firmware still accepts Root?

jbatonnet, 2025-04-30:
In the Wiki on this page, there's a section about official updates: https://thedju.notion.site/Installation-and-firmware-updates-1e145e4b6fd88023a4f5dc8e659a0468

jbatonnet, 2025-04-30:
Here is a test build with 2.4.0 support: https://github.com/jbatonnet/Rinkhals/releases/tag/20250430_01_test

closed 2025-05-07 by jbatonnet