jbatonnet/Rinkhals #157: Add support for the Kobra 3 Max

jbatonnet, 2025-04-30:


Myrkskog, 2025-05-05:
Does the firmware parody the kobra 3 aside from the build volume? Or no.

jbatonnet, 2025-05-05:
For now we don't have access to the K3M firmware. I guess it's very similar, but I cannot check

Myrkskog, 2025-05-05:
Got it. Hope they release it soon.

nikitax75, 2025-05-10:
Hi there, very interested in this one. My company has 4 of these with one half-broken that we use for testing & parts. We can do testing if that helps. 

I saw a Github where the guy managed to connect ADB on an older firmware, I'll try that to see if I can dump their latest firmware

jbatonnet, 2025-05-10:
That would be great, we're actively trying to have access to the SWU password for the K3M.
I suggest you join the Discord server, in the Rinkhals / kobra-3-max channel where we're discussing our progress!