jbatonnet/Rinkhals #158: Great mod for my K1 - with one request

BigToyBox, 2025-05-02:
This isn't really a "bug" but more of a request for the next version. 

Kobra sequence:
1. Heats to job bed temp and 170 nozzle
2. wipes nozzle
3. cools back to 140 and probes z
4. parks at chute and warms to job temps
5. poops (even if same filament is used)
6. wipes a second time
7. print starts

Why can't it just go to bed and extruder temps and do it's probe and wipes at the print temps?

basvd, 2025-05-03:
I think it follows this elaborate sequence to ensure the nozzle is clean before probing the bed. 170C is warm enough to soften the filament that might be stuck on the nozzle, allowing it to be wiped off. Then it cools to 140C to prevent filament ooze (note it presses down on the wiping area which keeps the nozzle closed).

Flushing the filament seems unnecessary, but it's difficult/impossible to reliably keep track of what filament is in the extruder between jobs. At the end of each job the filament is cut and retracted, and it's possible the user manually changes the filament around, which cannot be detected.

I have not experienced the second wipe you mention. For me it immediately starts printing after warming up to print temperature. I'm using a K3 though.

That being said, I don't know of a way to change this pre-print sequence. It might be baked into AC's klipper implementation.

BigToyBox, 2025-05-03:
Let’s hope the next AC version simply takes it up to first layer temp and
leaves it there for all the prep wipe steps. Retraction is there to handle
drips. It’s as if these programmers at the big companies have never had to
print the same part 20 times in a day.

By the way I tried the print queue in Mainsail and it showed an ACE error.

jbatonnet, 2025-05-07:
Closing this one as there's nothing that can be done to tweak those internals procedure for now

closed 2025-05-07 by jbatonnet