jbatonnet/Rinkhals #160: Macros are missing in Mainsail / Fluidd

jbatonnet, 2025-05-04:


CLNilix, 2025-06-01:
Hi Julian, thanks for opening this issue (i forgot to open an issue as you suggested on discord).
Here are some additional info (mainly copy and paste from discord):

it seems there are 7 macros from Anycubic in the printer.cfg of the KS1 by default (6 WIPE macros and 1 CANCEL_PRINT macro).
But i don't see any of these macros in Fluidd (the dashboard section "Macros" is empty).
In the macros settings section of fluidd, there are 7 "macros without category". If i click on this entry (7/7), it leads to an empty page.
I added some simple macros in the printer.custom file
```
[gcode_macro STEPPER_OFF]
description: schaltet alle Stepper Motoren aus
gcode:
    # disable all stepper motors
    M84

[gcode_macro COOL_DOWN]
description: schaltet alle heizungen aus
gcode:
    # turn off heatbed
    M140 S0
    # turn off hotend
    M104 S0

[gcode_macro FANS_OFF]
description: schaltet alle Ventilatoren aus
gcode:
    M106 P1 S0
    M106 P2 S0
    M106 P3 S0
```

but they don't appear in fluidd (neither in the dashboard nor in the settings).

jbatonnet, 2025-06-17:
Available in latest release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250617_01

closed 2025-06-17 by jbatonnet

CLNilix, 2025-07-06:
same with rinkhals 20250629_02; no macros and "old" fluidd version.