jbatonnet/Rinkhals #161: USB Ethernet support

alatum, 2025-05-04:
I was wondering if it was possible to enable USB ethernet support? It would help keep a much more stable connection. Not sure how easy it would be knowing drivers would probably be a big issue.

IguanaBob, 2025-05-04:
I was hoping for this as well. I did some investigation of the Kobra 3's OS over SSH and from what I can tell they include a very limited number of drivers, none of which allow USB-ethernet, and the kernel is so stripped down that my attempt to compile and load a driver was not successful. Based on that I suspect a full recompile of the kernel would be needed, which I suspect is way outside the scope of this or similar projects. However, I'm no pro in embedded Linux nor a developer for Rinkhals, so if anything I said is incorrect, someone please correct me.

jbatonnet, 2025-05-05:
Some tests have been done and someone on Discord managed to get it working at some point: https://discord.com/channels/1332498592539345069/1332500635857453106/1354471538006687764

Some manual configuration was needed, but it's surely possible to automate it in Rinkhals

jbatonnet, 2025-05-05:
Sorry for the message, I remember wrong. I thought someone was able to to it at some point but I was mistaken.

basvd, 2025-05-05:
From what I've seen the kernel has no built-in drivers or modules for USB ethernet, confirming what @IguanaBob also found. In theory the driver can be built as a module and loaded at runtime by Rinkhals, but someone will have to figure out how to compile for the stock kernel. Using Rinkhals' buildroot could be a good starting point.

As for the wifi connectivity problems, there are mods available to improve the antenna by replacing it with an external one. Maybe this will be a good alternative for some. (e.g. [KS1](https://www.makeronline.com/en/model/External%20Wifi%20Antenna%20for%20S1/157218.html), [K3](https://makerworld.com/en/models/1216792-anycubic-kobra-3-external-sma-wifi-antenna-mount#profileId-1232808))

andrewseago, 2025-05-15:
So I did a thing... on the KS1 and latest firmware. 

[ 1641.564241] ax88179_178a 1-1.4.3.3:1.0 eth1: register 'ax88179_178a' at usb-xhci-hcd.0.auto-1.4.3.3, ASIX AX88179 USB 3.0 Gigabit Ethernet, 00:05:1b:e1:fa:f6
[ 1642.612268] usb 1-1.4.3.4: new high-speed USB device number 12 using xhci-hcd
[ 1643.103253] usb 1-1.4.3.4: New USB device found, idVendor=2109, idProduct=0100, bcdDevice= 2.08
[ 1643.103278] usb 1-1.4.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1643.103287] usb 1-1.4.3.4: Product: USB 2.0 BILLBOARD             
[ 1643.103296] usb 1-1.4.3.4: Manufacturer: VIA Technologies Inc.         
[ 1643.103304] usb 1-1.4.3.4: SerialNumber: 0000000000000001
[ 1705.494716] ax88179_178a 1-1.4.3.3:1.0 eth1: ax88179 - Link status is: 1

eth1      Link encap:Ethernet  HWaddr 00:05:1B:E1:FA:F6  
          inet addr:10.0.1.210  Bcast:10.0.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3069 errors:0 dropped:129 overruns:0 frame:0
          TX packets:34 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1286590 (1.2 MiB)  TX bytes:5132 (5.0 KiB)

andrewseago, 2025-05-16:
root@Rockchip:/root# cat /sys/bus/usb/drivers/ax88179_178a/1-1.4.3.3:1.0/uevent 
DEVTYPE=usb_interface
DRIVER=ax88179_178a
PRODUCT=b95/1790/100
TYPE=255/255/0
INTERFACE=255/255/0
MODALIAS=usb:v0B95p1790d0100dcFFdscFFdp00icFFiscFFip00in00

And, just if anyone cares, eth0 is a built in Ethernet adapter in the Rockchip MCU. Someone with way too much time on their hands and a very steady soldering iron might be able to add an ethernet port to this thing without using USB. 

root@Rockchip:/root# cat /sys/bus/platform/devices/ffa80000.ethernet/uevent 
DRIVER=rk_gmac-dwmac
OF_NAME=ethernet
OF_FULLNAME=/ethernet@ffa80000
OF_COMPATIBLE_0=rockchip,rv1106-gmac
OF_COMPATIBLE_1=snps,dwmac-4.20a
OF_COMPATIBLE_N=2
OF_ALIAS_0=ethernet0
MODALIAS=of:NethernetT(null)Crockchip,rv1106-gmacCsnps,dwmac-4.20a

IguanaBob, 2025-05-19:
@andrewseago 

Question, did it work for you without a hub? Some of us on discord are trying to built/load modules for the same NIC but it won't work, but I don't have a hub to try.

IguanaBob, 2025-05-20:
So, a couple of us were trying 3.0 and 2.0 hubs, but no luck.

@andrewseago 
If at all possible could you share the drivers on the S1. Should be stored in /oem/usr/. I would like to try loading them on the K3M to see if it works.

Thanks.

IguanaBob, 2025-05-25:
All the chatter about this is on the Discord side. Just to update, Nikitax compiled CDC modules which are confirmed to load and work for CDC compatible NICs (AX88179, RPi g_ether, possibly others) on both Kobra 3 Max and Kobra 3. Eventually we should see this in the official build, but if you want to load it yourself for now the modules were posted on Discord.

gran8991, 2025-08-09:
Hello everyone,

I've been following this discussion with great interest as I'm trying to get a wired Ethernet connection working on my Kobra S1. I just want to confirm my plan before I proceed.

Here is my setup:

Printer: Anycubic Kobra S1

Stock Firmware: 2.5.6.0

Rinkhals Version: 20250725_01

Based on the documentation and this thread, here is what I plan to do:

Purchase a USB to Ethernet adapter with the ASIX AX88179 chipset (like the Amazon Basics model).

Add the following line to the /useremain/rinkhals/start-rinkhals.sh file, right after the log "Starting Rinkhals..." part:

udhcpc -i eth0

My question is: are these two steps all I need to do to get the Ethernet connection working? Or are there any other modifications or steps required that I might have missed?

Thank you for your amazing work on this project and for any guidance you can provide!

gran8991, 2025-08-12:
Can anyone tell me where I'm going wrong?

root@Rockchip:/root# cat /useremain/rinkhals/start-rinkhals.sh
#!/bin/sh

function log() {
    echo "${*}"
    echo "$(date): ${*}" >> /useremain/rinkhals/rinkhals.log
}

log
log "Starting Rinkhals..."

udhcpc -i eth1

CALLER=$(cat /proc/$PPID/cmdline)
log "Rinkhals startup was called from $CALLER ($PPID)"

if [ ! -f /useremain/rinkhals/.version ]; then
    log "Rinkhals is installed but no version is selected to start"
    exit 1
fi

if [ -f /mnt/udisk/.enable-rinkhals ] || [ -f /useremain/rinkhals/.enable-rinkhals ]; then
    log "Forcing Rinkhals startup with the .enable-rinkhals file"

    rm /mnt/udisk/.disable-rinkhals
    rm /useremain/rinkhals/.disable-rinkhals
fi

if [ ! -f /mnt/udisk/.disable-rinkhals ] && [ ! -f /useremain/rinkhals/.disable-rinkhals ]; then
    RINKHALS_VERSION=$(cat /useremain/rinkhals/.version)
    log "Rinkhals version $RINKHALS_VERSION selected"

    if [ ! -d /useremain/rinkhals/$RINKHALS_VERSION ]; then
        log "Rinkhals version $RINKHALS_VERSION does not exist"
        exit 1
    fi

    cd /useremain/rinkhals/$RINKHALS_VERSION

    chmod +x ./start.sh
    ./start.sh
fi

if [ -f /mnt/udisk/.disable-rinkhals ] || [ -f /useremain/rinkhals/.disable-rinkhals ]; then
    log "Rinkhals startup was stopped with the .disable-rinkhals file, trying to start ADB..."

    adbd &
    exit 1
fi

root@Rockchip:/root# lsusb
Bus 001 Device 001: ID 1d6b:0002
Bus 001 Device 007: ID 28e9:018a
Bus 001 Device 004: ID 13d3:784b
Bus 001 Device 002: ID 05e3:0610
Bus 002 Device 001: ID 1d6b:0003
Bus 001 Device 008: ID 0b95:1790
Bus 001 Device 003: ID 05e3:0610

[  206.432703] usb 1-1.4: new high-speed USB device number 9 using xhci-hcd
[  206.618252] [ac]usb_parse_endpoint:272,size=0,----
[  206.667895] [ac]usb_parse_endpoint:272,size=0,----
[  206.716548] [ac]usb_parse_endpoint:272,size=0,----

root@Rockchip:/root# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 0A:31:E2:7D:3D:FD
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:45

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:200 errors:0 dropped:0 overruns:0 frame:0
          TX packets:200 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:34334 (33.5 KiB)  TX bytes:34334 (33.5 KiB)

p2p0      Link encap:Ethernet  HWaddr 2A:F5:2B:B7:48:29
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr 28:F5:2B:B7:48:29
          inet addr:192.168.1.56  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:945 errors:0 dropped:112 overruns:0 frame:0
          TX packets:267 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:399751 (390.3 KiB)  TX bytes:142008 (138.6 KiB)

Eszartek, 2025-08-19:
I have a Ugreen 3 port USB hub with Gig Ethernet adapter that uses AX88179.
It works great, but the drivers are not included in Rinkhals yet. You need to go to the #development channel on Discord to download the kernel modules and installer script.

Eszartek, 2025-08-19:
I editied /useremain/rinkhals/start-rinkhals.sh to load the modules and set a static ip, but you can use dhcp if you prefer:

```
log "Starting Rinkhals..."
IP_ADDRESS="192.168.0.80"
REMOTE_DRIVER_DIR="/useremain/usb_net_drivers"
insmod $REMOTE_DRIVER_DIR/usbnet.ko
insmod $REMOTE_DRIVER_DIR/cdc_ether.ko
insmod $REMOTE_DRIVER_DIR/cdc_ncm.ko
insmod $REMOTE_DRIVER_DIR/mdio-bitbang.ko
insmod $REMOTE_DRIVER_DIR/ax88796.ko
insmod $REMOTE_DRIVER_DIR/asix.ko
ifconfig eth1 $IP_ADDRESS up
#udhcpc -i eth1
```

StephenGoodall, 2025-08-20:
Thanks @Eszartek - that's brilliant!

One more dumb question - does it only work with the slicer when using LAN mode? Does wifi always need to be on for the cloud mode to work?

Devirex, 2025-09-15:
> Awesome. Are you on discord? That's were we have been working on a bundled installer/swu and would love to get your driver added to it. https://discord.com/channels/1332498592539345069/1375164318454251663

Yes i have an account, but my app (or me) cannot handle the link 😄