jbatonnet/Rinkhals #162: Hard disk full, Delete Rinkhals files ?

Joerg1309, 2025-05-06:
I have tried to update the firmware.

However, this does not work because it complains that the hard drive is full. How can I remove the FILES FROM RINKHALS COMPLETELY ? only disabling does not work. or is there another solution ?

[image]

jbatonnet, 2025-05-06:
There's a section about that on the main page: https://github.com/jbatonnet/Rinkhals?tab=readme-ov-file#how-to-uninstall-rinkhals
You'll need to use a SSH tool to get SSH access (described on the main page as well), or use ADB and remove Rinkhals directory.

jbatonnet, 2025-05-06:
Next feature release will come with a tool to help in this situation but it's not ready yet.

jbatonnet, 2025-05-06:
I added a section in the FAQ over there: https://thedju.notion.site/FAQ-1e145e4b6fd880038ce2dd0f7615792f#1eb45e4b6fd88035901ac50f28cb5c97

You'll need a SSH client for Mac. Once connected, you can use:
`rm -rf /useremain/rinkhals`

closed 2025-05-06 by jbatonnet

Joerg1309, 2025-05-06:
Ok thanks , but they will a User and passwort for login

Antworten an: jbatonnet/Rinkhals ***@***.***>
Datum: Dienstag, 6. Mai 2025 um 15:45
An: jbatonnet/Rinkhals ***@***.***>
Cc: Jörg Elbe ***@***.***>, Author ***@***.***>

Ucsus, 2025-05-08:
You can do this through the WinSCP program - new connection, IP of your printer, login "root", password "rockchip", go to the directory /useremain and delete the folder "rinkhals"

Joerg1309, 2025-05-08:
Thanks, I will try it. I become a new Mainboard from Anycubic, The printer says Harddisk full, but I delete all Files.
The say 5,2 Gb are Used from 5,9Gb.  I put a Usb in the Port and the Stick was very Hot and has Damage the Housing from the stick.