jbatonnet/Rinkhals #163: Can't set Z axis offset

tom-321, 2025-05-06:
There ist just an empty field in mainsail: 

[image]

jbatonnet, 2025-05-06:
Make sure you're running last Rinkhals version. Mainsail had a bug about that that was fixed recently.
Mine with latest version has this module

[image]

tom-321, 2025-05-06:
Hi, is this while printing or in IDLE ? 'cause for me it disappears when I start a print,
and that's exactly when I want to adjust the Z offset...

jbatonnet, 2025-05-07:
Idle, it may very well not be possible to adjust the zoffset during print on our printers. I'll keep it for now but I remember reading about that somewhere.
Try doing that from the screen itself for now

tom-321, 2025-05-07:
Yeah, I've tried to find a option on the screen on the kobra s1, it doesn't seem like there's something like that...

basvd, 2025-05-09:
It used to be possible to adjust the z-offset via Mainsail during a print. I know because I've done it this way myself (not trusting the screen UI method). I don't remember which version exactly, but I think it was 2.3.5.3 on the K3.

shoikan, 2025-06-12:
Using Rinkhals 20250531_01 on KS1C with firmware v2.5.2.3. I can see the Z-Offset controls in Mainsail, both before and during the print. However they do not appear to be actually doing anything as far as the GUI is concerned. If I press any of the adjustments (+ or -) while homed, but not printing, I can hear the printer making adjustments. But the Z-Offset indicator (above the +/- buttons) stays at 0. I can also not see any impact on the reported Z positioning of the toolhead. During printing, I can't see if this is making any difference either. I don't seem to see any effect on the printing height.