jbatonnet/Rinkhals #164: Current print job isn't diplayed on the printer's screen

tom-321, 2025-05-06:
If I start a print in the web interface or in orcaslicer, it won't show up on the printers display.

jbatonnet, 2025-05-06:
You need LAN mode to be enabled for prints to show on screen

closed 2025-05-06 by jbatonnet

tom-321, 2025-05-06:
Perfekt, work's 👍
Thanks 🙌

tom-321, 2025-05-07:
Since I've enabled LAN mode I get this screen after restarting the printer:

[image]

tom-321, 2025-05-07:
In my browser it look's like this:

[image]

tom-321, 2025-05-08:
I found a "workaround" 
if I start the printer, it seems to hassle a bit with the wifi connection (I can't get a consistent ping) 
But if I instantly restart the printer after the start (complete power off the main switch) it establishes a Wi-Fi connection and I can connect via orcaslicer...

tom-321, 2025-05-08:
Oh, I saw there was a update 2 days ago, I'll try it out, maybe it's already working

tom-321, 2025-05-08:
Ok no I can't get any wifi connection, does anyone else have this problem ?
Mabe it my wifi..