jbatonnet/Rinkhals #165: Request for Comprehensive Wiki

n3roGit, 2025-05-06:
Hello Rinkhals,

Firstly, thank you for your hard work on the Rinkhals project. As a new user, I find the software promising but challenging to navigate without comprehensive documentation.

I would like to request the creation of a detailed wiki or user guide that covers the following areas:

Configuration Settings: Detailed explanations of configuration files, their locations, and how to modify them for various use cases.

Web UI Access: Instructions on enabling and accessing the Web UI, including default ports and any necessary prerequisites.

OrcaSlicer Integration: Step-by-step guidance on setting up a local printer connection via OrcaSlicer, including any required plugins or settings.

Having such documentation would greatly enhance the user experience, especially for newcomers trying to leverage Rinkhals effectively.

Thank you for considering this request.

Meowcat285, 2025-05-12:
@n3roGit Just to let you know, we've moved over to using MkDocs for the wiki now: [https://jbatonnet.github.io/Rinkhals/](https://jbatonnet.github.io/Rinkhals/)

jbatonnet, 2025-05-18:
All of that has been documented there: https://jbatonnet.github.io/Rinkhals/Rinkhals/
More documentation for the rest is coming

closed 2025-05-18 by jbatonnet