jbatonnet/Rinkhals #166: Exclude Objects not working on an S1

VigilantGardener, 2025-05-06:
**Describe the bug**
Exclude Objects is not working on an S1 running Rinkhals  20250424 and firmware 2.5.2.3
The Exclude Objects controls were not present in Mainsail or Fluidd

**To Reproduce**
Steps to reproduce the behavior:
1. These lines were added to the moonraker.custom config file
[file_manager]
enable_object_processing: True
2.  Afterward the printer was rebooted and it was confirmed that those lines were added to the moonraker.generated config file
3.  Using Orca Slicer the options for "Label Objects" and "Exclude Objects" were set on the print profile
4. Sent print to printer

**Logs / Additional information**
The factory printer config file already included the exclude_objects module, no changes were made to the printer config files, just moonraker.
No errors are logged in the Console in Mainsail.

VigilantGardener, 2025-05-06:
The Klipper documentation includes details on how to manually send the exclude object command.  So as a test that was done.  It seems like the command was accepted, but the object did not cease printing.

The print was allowed to complete without any interference, and then once it was done a debug bundle was gathered from the machine.

[debug-bundle_2516009911_19700101-020003.zip](https://github.com/user-attachments/files/20070775/debug-bundle_2516009911_19700101-020003.zip)

[image]

rainbyte, 2025-05-10:
I noticed this is also happening on Kobra 3 Combo with fw 2.4.0 and rinkhals 20250506_01.

Same options were applied, but exclude objects is missing on screen and fluidd UI.

Art531, 2025-05-13:
For now "Exclude_objects" feature is supported only with gcodes sliced in Anycubic Slicer Next, they have added some proprietary format for that. Orca slicer is not providing this feature, and it is not firmware-dependent, it is gcode-dependent issue.

VigilantGardener, 2025-05-13:
> For now "Exclude_objects" feature is supported only with gcodes sliced in Anycubic Slicer Next, they have added some proprietary format for that. Orca slicer is not providing this feature, and it is not firmware-dependent, it is gcode-dependent issue.

Can you be more specific?

I'm looking at the GCode files and the only difference that really stands out is that the G1 command to adjust Z height is issued first in AnyCubic Next Slicer.  It does include the comment of "For Object Exclusion".  Maybe that's enough of a difference to make a difference.

[image]
[image]

VigilantGardener, 2025-05-13:
This is a GCode file generated by AnyCubic Next Slicer.  When printing exclude objects is working in the Workbench of Next Slicer, but there are no options for exclude objects in Mainsail or Fluidd.

[GCode File for ticket.txt](https://github.com/user-attachments/files/20196120/GCode.File.for.ticket.txt)

VigilantGardener, 2025-05-16:
A print sent from Orca Slicer was able to call the object exclude function from the screen on the S1, the exclude function also works properly.

However, the exclude object commands are still not available in mainsail or Fluidd with this change.

The only change made in Orca was to add a 2nd thumbnail to the printer's configuration.  See attached image.

Additionally, the thumbnail previews are now working in Mainsail.  See image

[image]
[image]

VigilantGardener, 2025-05-18:
I believe that the object exclude commands are being issued to the printer via MQTT.  Maybe that's why it isn't working in Mainsail/Fluidd?

Method used to come to this conclusion
1.  Started a print via AnyCubic Slicer , in LAN only mode
2. Used the Slicer workbench to send an object exclude command
3. Wireshark was used to capture traffic between PC and printer
4. At the time the exclude object command was sent, packets destined to port 9883 were transmitted from the PC to the printer.
5. Packets were encrypted, TLS v1.2, so the payload could not be verified
6. Assumptively, the typical object exclude functionality has been superseded by this MQTT arrangement.

martinbogo, 2026-03-30:
Okay, quick update.  I fixed this, and will add the fix into the develop branch for next release down the chain after this one. 

I started by reproducing the problem in both Mainsail and Fluidd, and confirmed that there were really two separate issues mixed together. First, the compatibility layer was not exposing exclude_object consistently enough for clients, which affected UI behavior. Second, **even when exclusion state appeared in the UI**, the printer could still continue physically printing that object in some cases because of the way things are handed under the hood.

From there, I traced the command path all the way through Moonraker compatibility into GoKlipper. GoKlipper has native object exclusion endpoints (_exclude_object/get_objects and exclude_object/set_objects_), and set_objects only applies correctly when the payload uses the exclude_objects key. So, it was a bit tricky.

My fix intercepts when Mainsail/Fluidd sends EXCLUDE_OBJECT NAME=..., Rinkhals now routes that to GoKlipper’s native exclusion API directly, instead of relying on compatibility layer.  You have to hook directly into the way GoKlipper wants things to be.

So at this point, I have object exclusion working with the tested setup on firmware 2.7.0.9 on my KS1 using the develop branch, and because the fix is in backend command translation, it applies to **_both_** Mainsail and Fluidd rather than being frontend-specific.