jbatonnet/Rinkhals #167: Error with different nozzle sizes

EikeKre, 2025-05-07:
**Describe the bug**
After uploading a gcode with a 0.6 nozzle the print is immediately stopped without an error in the frontend. Tried to set the nozzle_diameter in `/useremain/home/rinkhals/printer_data/printer.custom.cfg` below `[extruder]` Section which didn't work after reboot. What solved this issue was to set the diameter in the printers menu itself.

**To Reproduce**
Steps to reproduce the behavior:
1. Upload gcode with a 0.6 nozzle in the printer settings of OrcaSlicer (AnycubicSlicerNext)
2. Print is immediately stopped

**Logs / Additional information**
/useremain/rinkhals/.current/logs/app-moonraker.log
```
[application.py:prepare()] - Upload Request Received from 192.168.10.85
Content-Type: multipart/form-data; boundary=------------------------54ca5352c8fd6fa1
[application.py:post()] - Processing Uploaded File: testfile.gcode
[shell_command.py:pipe_data_received()] - INFO:metadata:Processor preprocess_cancellation is disabled

[shell_command.py:_check_proc_success()] - Command (/useremain/rinkhals/20250424_01/home/rinkhals/apps/40-moonraker/bin/python /useremain/rinkhals/20250424
_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/file_manager/metadata.py -c /tmp/metacfg-o9ig5rw2.json) successfully finished
[klippy_apis.py:start_print()] - Requesting Job Start, filename = Plant_Pot_resized_holes_cut_PLA_0.3_7h8m.gcode
[kobra.py:mqtt_print_file()] - Trying to print Plant_Pot_resized_holes_cut_PLA_0.3_7h8m.gcode using MQTT...
[kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_auto_leveling" => ""
[kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_vibration_compensation" => ""
[kobra.py:shell()] - Shell ". /useremain/rinkhals/.current/tools.sh && get_app_property 40-moonraker mqtt_print_flow_calibration" => ""
[kobra.py:mqtt_on_message()] - Received MQTT print state: failed
[kobra.py:mqtt_print_file()] - Error while trying to print: The nozzle size of the gcode file is inconsistent with the nozzle size of the device (code 1023
6)
[application.py:log_request()] - 201 POST /api/files/local (192.168.10.85) [_TRUSTED_USER_] 6233.47ms
[proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 4.34 seconds, total blocked count: 2
[proc_stats.py:log_last_stats()] -
```

jbatonnet, 2025-05-07:
The error "The nozzle size of the gcode file is inconsistent with the nozzle size of the device (code 10236)" comes from Anycubic binaries themselves. So the right thing to do is to do it from the printer screen UI.
Tweaking the printer config is not recommended for this printer: https://www.notion.so/thedju/Printer-configuration-1e145e4b6fd880bfaee4c4d6e42421b2

closed 2025-05-07 by jbatonnet

EikeKre, 2025-05-07:
Thank you. Good to know! Maybe worth a note in the readme?