jbatonnet/Rinkhals #170: Send messages to external MQTT server

Kixel, 2025-05-09:
I want to send messages to another external mqtt server (so my homeassistant server can do stuff with it, i.e. trigger camera etc.) and found this: https://klipper.discourse.group/t/m118-publish-to-mqtt/1123 . 
I tried to configure the external mqtt server ( https://moonraker.readthedocs.io/en/latest/configuration/#mqtt ) but got no results. Logs show "unknown command". I added the suggested macro to the printer.custom.conf, as well as the mqtt: config part.
Any insights (or even better approaches) for this?

basvd, 2025-05-09:
What exactly did you add and what did the logs say exactly? In theory it should be possible to trigger a command from Gcode, see this page about Spoolman for example: https://thedju.notion.site/Spoolman-support-1ee45e4b6fd880b1869de8735ed85052

Do note that you have to use double {{ brackets instead of the single ones that normal Klipper expects.

Kixel, 2025-05-09:
I added this to printer.custom.cfg, pretty much as described in https://klipper.discourse.group/t/m118-publish-to-mqtt/1123/5 

`[mqtt]
address: i.redacted.the.ip
port: 1883
client_id: kobras1
enable_tls: False
username: mqttuser
password: redacted
enable_moonraker_api: False
instance_name: kobras1

[gcode_macro PUBLISH_ALERT]
gcode:
  {% set data = params.PAYLOAD|default(None) %}
  {action_call_remote_method("publish_mqtt_topic",
                             topic="klipper/gcode",
                             payload=data,
                             qos=0,
                             retain=False,
                             use_prefix=True)}`

basvd, 2025-05-09:
Maybe if you use double `{{` brackets around the `action_call_remote_method` part it will work. I also believe that custom gcode that this klipper implementation is able to parse is limited to the format `<letter><numbers> <letter>=<param>`, e.g. `M999 A=9`, so you might have to set up an extra macro for that gcode (again, see the Spoolman example).

basvd, 2025-05-09:
Just to be sure, you did add the `[mqtt]` section to the Moonraker config, right? It's not supposed to be in printer.cfg

Kixel, 2025-05-09:
> Just to be sure, you did add the `[mqtt]` section to the Moonraker config, right? It's not supposed to be in printer.cfg

I actually added both (the mqtt part and the macro) to both printer.custom.conf and moonraker.custom.conf.
But from what i can tell, neither part is actually added to any *.generated.conf .

Double brackets didn't change a thing.

jbatonnet, 2025-05-09:
The printer generated file is generated during boot, so you need a reboot.
The moonraker generated file is built during the Moonraker app startup, so restarting it or rebooting will work.

But yeah, I highly doubt the mqtt section in printer will have an effect and maybe will prevent the printer from starting

Kixel, 2025-05-09:
Alright, i made some progress. I added the mqtt part to moonraker.custom.conf. After a complete reboot, these settings are added to the actual moonraker.conf. I changed the settings of mqtt a bit so it publishes the print time every so often. This now works fine. It also updates the moonraker server status for some reason (in mqtt).

What doesn't yet work is publishing an actual message to mqtt. If I try to add the sample macro to the printer.custom.conf, the printer won't boot (and i have to recover with a backup printer.custom.conf and a .enable-rinkhals file as described in the excellent documentation for this: https://thedju.notion.site/Printer-configuration-1e145e4b6fd880bfaee4c4d6e42421b2 ). 
Any insights on this?

basvd, 2025-05-10:
Have you tried a printer config like this, and use `M555 P=...` in your gcode instead of `PUBLISH_ALERT`?
```
[gcode_macro PUBLISH_ALERT]
gcode:
  {% set data = params.PAYLOAD|default(None) %}
  {{action_call_remote_method("publish_mqtt_topic",
                             topic="klipper/gcode",
                             payload=data,
                             qos=0,
                             retain=False,
                             use_prefix=True)}}

[gcode_macro M555]
gcode:
    PUBLISH_ALERT PAYLOAD={{ params.P }}
```

Kixel, 2025-05-10:
I tried adding what you suggested here to my printer.custom.cfg and it 11407's aka can't boot and i have to recover with the usb drive. Didn't even get to try any gcode. Did this work on your end?

basvd, 2025-05-10:
No it doesn't work for me either. Looking into it more, I found that the template language is different from vanilla klipper and a bit more limited (pongo2). The config that worked for me, that is no boot error, is the following:

```
[gcode_macro PUBLISH_ALERT]
gcode:
  {{ action_call_remote_method("publish_mqtt_topic", params) }}

[gcode_macro M555]
gcode:
    PUBLISH_ALERT PAYLOAD={{ params.P|default:None }} TOPIC="klipper/gcode", QOS=0, RETAIN=False, USE_PREFIX=True
```

It should be functionally equivalent to what I posted before. The reason why it's done this way is to get around the fact that pongo2 doesn't support named parameters, which is what `action_call_remote_method` normally requires. Instead, the M555 helper code is used to set all the parameters by name that the remote method requires. Then that whole list is passed as a single `params` in the actual `PUBLISH_ALERT` command.

I haven't tested further. The parameters being uppercase might cause it to not work, and I am also not entirely sure about the syntax of the values. I just kept them as they were originally.

Hope this helps.

Kixel, 2025-05-10:
alright, i'll try this tomorrow (when i don't have a print running anymore). will post a full overview of my setup here as well once it works - just in case anyone else needs/wants this.

Kixel, 2025-05-11:
Ok, with your macros added it doesn't fail boot-up, but the moonraker.log shows this:
```
2025-05-11 17:58:07,345 [klippy_connection.py:_execute_method()] - Error running remote method: publish_mqtt_topic
Traceback (most recent call last):
  File "/useremain/rinkhals/20250506_01/home/rinkhals/apps/40-moonraker/moonraker/moonraker/components/klippy_connection.py", line 561, in _execute_method
    ret = self.remote_methods[method_name](**kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: MQTTClient._publish_from_klipper() got an unexpected keyword argument 'PAYLOAD'

```

basvd, 2025-05-11:
Looks like the uppercase arguments might be the problem. You could try making them lowercase in the gcode definition, but I'm not sure if go-klipper will accept them as valid. If that doesn't work then it will require some programming to make moonraker case-insensitive.

jbatonnet, 2025-05-26:
Since this one is about trying Gcode for Anycubic, I'll close it and I suggest joining and discussing that on the Rinkhals Discord!

closed 2025-05-26 by jbatonnet