jbatonnet/Rinkhals #173: Rinkhals Monitor App not started

Toolmaster30, 2025-05-10:
I'm using Rinkhals on my Kobra 3 with firmware 2.4.0. Unfortunately, the monitoring app won't launch. Is there a solution?
Regards, Georg

jbatonnet, 2025-05-10:
What do you mean by won't launch? Is it enabled but not starting from the UI or on reboot?

The monitoring app behavior is described here: https://jbatonnet.github.io/Rinkhals/Rinkhals/monitoring-app/
It does not provide any interface, just a log and it publishes to the internal MQTT server if LAN mode is enabled.

Toolmaster30, 2025-05-11:
The app is enabled on the printer. But when I try to start it using the "Start" button, it starts briefly. It displays usage and memory usage, but then immediately closes again.

jbatonnet, 2025-05-11:
Okay, please generate a debug bundle as detailed on the home page to collect your printer logs

Toolmaster30, 2025-05-12:
`-------------------- Log Start | Mon May 12 04:37:44 2025 --------------------
platform: Linux-5.10.160-armv7l-with-libc0
data_path: /userdata/app/gk/printer_data
is_default_data_path: True
config_file: /userdata/app/gk/printer_data/config/moonraker.generated.conf
backup_config: /userdata/app/gk/printer_data/config/.moonraker.generated.conf.bkp
startup_warnings: []
verbose: False
debug: False
asyncio_debug: False
is_backup_config: False
is_python_package: False
instance_uuid: de34c0ac73784ba38156825c58ad0b1d
unix_socket_path: /userdata/app/gk/printer_data/comms/moonraker.sock
software_version: ?
log_file: /userdata/app/gk/printer_data/logs/moonraker.log
python_version: 3.11.6 (main, Feb  9 2025, 23:27:30) [GCC 11.4.0]
launch_args: /useremain/rinkhals/20250506_01-2/home/rinkhals/apps/40-moonraker/bin/python ./moonraker/moonraker/moonraker.py -c /userdata/app/gk/printer_data/config/moonraker.generated.conf
msgspec_enabled: False
uvloop_enabled: False

#################### Moonraker Configuration ####################

[server]
host = 0.0.0.0
port = 7125
klippy_uds_address = /tmp/unix_uds1
max_upload_size = 1024

[machine]
provider = none

[authorization]
cors_domains = 
	*://my.mainsail.xyz
	*://app.fluidd.xyz
	*://*.local
	*://*.lan
trusted_clients = 
	10.0.0.0/8
	127.0.0.0/8
	169.254.0.0/16
	172.16.0.0/12
	192.168.0.0/16
	FE80::/10
	::1/128

[octoprint_compat]

[history]

[database]

[kobra]

#################################################################
All Configuration Files:
/userdata/app/gk/printer_data/config/moonraker.generated.conf
#################################################################
Loading Sqlite database provider. Sqlite Version: 3.40.1
Unsafe Shutdown Count: 9

System Info:

***python***
  version: (3, 11, 6, 'final', 0)
  version_string: 3.11.6 (main, Feb  9 2025, 23:27:30) [GCC 11.4.0]

***cpu_info***
  cpu_count: 1
[... 580 lines cut ...]
1970-01-01T00:00:19.686Z INFO project/Webhooks.go:384 webhooks client 642140b5-b9ff-458c-b36b-1da1417baaaf: New connection
1970-01-01T00:00:20.433Z DEBUG project/filament_hub.go:1007 filament_present is 1print_state is 0pre_cleaning_pipe is falsepre_refilling is falsefilamentIsEmpty is true
1970-01-01T00:00:21.649Z INFO project/Webhooks.go:384 webhooks client 72c814b6-1523-4d4d-adcf-c6eec35b7fd4: New connection
1970-01-01T00:00:21.650Z INFO project/Webhooks.go:384 webhooks client 7fc18316-c0cf-4bd1-b719-fda77e454d8c: New connection
1970-01-01T00:00:22.608Z INFO project/Webhooks.go:384 webhooks client 72c814b6-1523-4d4d-adcf-c6eec35b7fd4: Disconnected
1970-01-01T00:00:22.609Z INFO project/Webhooks.go:384 webhooks client 7fc18316-c0cf-4bd1-b719-fda77e454d8c: Disconnected
2025-05-12T04:40:16.828Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.1: no such file or directory
2025-05-12T04:40:16.829Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.2: no such file or directory
2025-05-12T04:40:16.829Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.3: no such file or directory
2025-05-12T04:40:16.829Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.4: no such file or directory
2025-05-12T04:40:16.831Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.5: no such file or directory
2025-05-12T04:40:16.831Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.6: no such file or directory
2025-05-12T04:40:16.837Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.7: no such file or directory
2025-05-12T04:40:16.837Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.8: no such file or directory
2025-05-12T04:40:16.838Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.9: no such file or directory
2025-05-12T04:40:16.838Z ERROR project/extras_virtual_sdcard.go:511 open /ac_app/record.10: no such file or directory
2025-05-12T04:40:36.599Z INFO project/Webhooks.go:384 webhooks client d416a878-7ff1-4795-abd2-98de743f54ec: New connection
2025-05-12T04:40:38.379Z INFO project/Webhooks.go:686 webhooks: registering remote method 'shutdown_machine' for connection id: 14036536
2025-05-12T04:40:38.408Z INFO project/Webhooks.go:686 webhooks: registering remote method 'reboot_machine' for connection id: 14036880
2025-05-12T04:40:38.514Z INFO project/Webhooks.go:686 webhooks: registering remote method 'pause_job_queue' for connection id: 14037392
2025-05-12T04:40:38.531Z INFO project/Webhooks.go:686 webhooks: registering remote method 'start_job_queue' for connection id: 14037552
2025-05-12T04:40:38.536Z INFO project/Webhooks.go:686 webhooks: registering remote method 'set_device_power' for connection id: 14037712
`

Good morning, these are my two log files.

jbatonnet, 2025-05-12:
A full debug bundle is needed, please refer to the home page instructions on how to create one. Thank you!

Toolmaster30, 2025-05-15:
I've uploaded the entire debug log file here. Hope it helps.

[debug-bundle_2407758264_20250515-060618.zip](https://github.com/user-attachments/files/20220557/debug-bundle_2407758264_20250515-060618.zip)

jbatonnet, 2025-05-16:
Based on the logs, the monitor were able to start on your printer, but it is now disabled.

```
2025/05/14 18:51:41 Connecting to MQTT broker at 127.0.0.1:2883
2025/05/14 18:51:41 Published Home Assistant discovery topic for device 372e61d7d25a307424c55ff9f0897bf5
2025/05/14 18:51:41 Found process mjpg-streamer with PID 915
2025/05/14 18:51:41 Found process K3SysUi with PID 709
2025/05/14 18:51:41 Found process gkapi with PID 708
2025/05/14 18:51:42 Found process Moonraker with PID 1271
2025/05/14 18:51:42 Found process Rinkhals monitor with PID 2620
2025/05/14 18:51:42 Found process gklib with PID 697
2025/05/14 18:51:44 Memory Usage: 106 MB / 212 MB, CPU Usage: 91.96141479099683%, CPU Load: 9.84
```

```
Skipped rinkhals-monitor (/useremain/rinkhals/20250514_01/home/rinkhals/apps/rinkhals-monitor) as it is not enabled
```

closed 2025-05-23 by jbatonnet