jbatonnet/Rinkhals #174: cannot print says its needs ressonnace

supersailor89, 2025-05-12:
i cannot print, it states that it needs the ressonance completed, and when i try to run the resonnance test the printer does not do anything.

bmorezcunee, 2025-05-12:
> i cannot print, it states that it needs the ressonance completed, and when i try to run the resonnance test the printer does not do anything.

I know this will sound unhelpful but Ive had similar weird issues.  Try these troubleshooting steps...

First thing I would do is turn on LAN MODE. Do not use the Anycubic App on your phone. It causes unnecessary problems.

Clear out old files stored on the printers local storage ( Old print jobs, time lapse videos, etc )

Then reboot the printer and run a full calibration. Try your print again after you do those things. I hope that helps.

jbatonnet, 2025-05-12:
Hi, thank you for your feedback. This issue is not directly related to Rinkhals but to Anycubic apps.
You can always switch to stock firmware from the screen to test the difference if needed.
Our Discord server has a section to discuss general Kobra related 3D printing topics, feel free to join and discuss that with the community!
I'm closing this issue for now!

closed 2025-05-12 by jbatonnet