jbatonnet/Rinkhals #175: Start G-CODE not working

masgar, 2025-05-12:
I want to force WIPE_NOZZLE and BED_MESH_CALIBRATE to run before every print, so I modified the start G-code in Orca like this:

G28  
G1 X47 F12000  
G1 Y276 F12000  
M140 S100  
M109 S190  
WIPE_ENTER  
WIPE_NOZZLE  
WIPE_STOP  
M109 S140  
WIPE_EXIT  
BED_MESH_CALIBRATE  
G1 X47 F12000  
G1 Y276 F12000  
G9111 bedTemp=55 extruderTemp=196  
M117  
[.....]
Here's what happens:

The printer performs homing (G28)

Sets bed temperature to 100°C

Heats the nozzle to 190°C (waiting to reach temp)

Then skips all the WIPE_xxx macros

Heats the nozzle to 140°C

Continues with the rest of the print

However, when I run the same WIPE_xxx commands directly via the Fluidd console, they execute correctly.

The commands are present in the exported/uploaded G-code file. I’ve tried starting the print from Orca and also from a USB key—same result: the WIPE_ commands are ignored.

Any suggestions would be appreciated.

Printer: Bambu Lab KS1
Firmware: Latest Rinkhals

mircsicz, 2025-05-12:
You sure about who produced your printer? 🤣

jbatonnet, 2025-05-12:
Hi, Rinkhals only interferes with the gcode when a single BED_MESH_CALIBRATE instruction is sent. Otherwise it is GoKlipper running the gcode and there's nothing that can easily be done in this case.
I suggest you join the Discord server to discuss what you want to achieve in the "3D printing" section, many people like you are playing with those printers!

closed 2025-05-12 by jbatonnet