jbatonnet/Rinkhals #176: "Save config and restart" does not restart GoKlipper in Moonraker

jbatonnet, 2025-05-12:


xlordxcheater, 2025-05-30:
Same here. The Save_config does nothing neither the FW restart. The only way i can see the changes in the heightmap and the machine files is by restarting the printer itself.

Kobra 3 2.4.0 fw
Rinkhals (Latest release-- not prerelease)

jbatonnet, 2025-06-13:
I was able to check this issue including my last changes. The SAVE_CONFIG is applied and config is saved including in the mutable files.
Despite the "& RESTART" part, the only gcode sent is SAVE_CONFIG.

I can confirm this command is received by GoKlipper and applies properly
[image]

But there's a bug left where Fluidd/Mainsail doesn't see the saved config. I think it comes from GoKlipper itself not clearing the pending changes flag in config save.
A GoKlipper restart might be needed, but I'm not sure how to do that yet.

closed 2025-06-18 by jbatonnet