jbatonnet/Rinkhals #180: Rinkhals 20250514_01

jstump7194, 2025-05-15:
Trying to install Rinkhals on my new s1 machine it is updated to 2.5.2.3. When I plug USB drive in with current (update.swu) Rinkhals interface comes up, I try to install but it stalls out at 98% every time. I even tried previous 2 versions and got the same result, stalled at 97-98%

slaapliedje, 2025-05-15:
Use the update-ks1.swu instead of the installer.  It'll do as the instructions states, beep twice and reboot.  If you then copy over the installer, it'll show the version is installed.

greenyellow-ai, 2025-05-15:
I had the same issue as OP and took a chance that the Kobra 3 would behave the same.  I had already downgraded my mfr firmware to 2.3.9.3 in hopes that Rinkhals would install that way.  I was just about to post with a debug dump when I saw this.

The update only version of 20250514_01 did not update the printer, or maybe I shut it off before it actually unpacked and updated.

I put the version just before that on, since it knew how to work with firmware 2.3.9.3.  Then I upgraded Rinkhals to the latest version.  I have an Elegoo printer and I don't miss much about it, but I did miss its web interface.  Now I have pretty much that same web interface on my Kobra 3.

I'll update the mfr firmware later tonight but just wanted to post with a Kobra 3 datapoint for a new install with the current version.

jstump7194, 2025-05-15:
I ended up having to go to the first version for 2.5.2.3 and update to current from there. Also used the update-ks1.swu. Maybe it was just a fluke or operator error. Maybe this will help someone else. Cheers happy printing!

closed 2025-05-15 by jstump7194