jbatonnet/Rinkhals #181: modificar hostname

adrianlibonatti, 2025-05-16:
Hi

Is there any way to change the hostname? I've already tried using the hostname command and modifying the configuration file in SSH, but it asks for admin privileges. I tried running sudo and su, but neither is installed...
I have two printers, a K3 and an S1. They both have the same hostname, which is a bit confusing.

Whatever the answer, thank you...

adrianlibonatti, 2025-05-21:
Thank you for your reply. I'll be patiently awaiting your changes. Best regards.

jbatonnet, 2025-05-28:
Okay I've been sleeping on this one for quite some time. I've been debating adding custom startup script or using apps. Apps are already a valid way to run any script on startup.
I think the most elegant way in this setup is to make yourself a custom user app to run exactly what you want.

You would have to:
- Create a directory /useremain/home/rinkhals/apps/myapp
- Inside, create 2 files app.json and app.sh

app.json
```
{
    "$version": "1",

    "name": "My app",
    "description": "My awesome app with cool things",
    "version": "1"
}
```

app.sh
```
. /useremain/rinkhals/.current/tools.sh

status() {
    report_status $APP_STATUS_STOPPED
}
start() {
    # Your code, changing hostname, ...
}
stop() {
    exit 0
}

case "$1" in
    status)
        status
        ;;
    start)
        start
        ;;
    stop)
        stop
        ;;
    *)
        echo "Usage: $0 {status|start|stop}" >&2
        exit 1
        ;;
esac
```

Then you can enable it using the Touch UI and it will run on each startup.

Have a look at the rinkhals-telemetry app: https://github.com/jbatonnet/Rinkhals/tree/20250517_01/files/4-apps/home/rinkhals/apps/rinkhals-telemetry

closed 2025-06-08 by jbatonnet