jbatonnet/Rinkhals #182: Add moonraker-timelapse to base pack

rowcount, 2025-05-17:
I was try to adding it manually, timelapse component initialize, but when I adding [include timelapse.cfg] to server.custom config - klipper is down

rowcount, 2025-05-17:
[gklib (5).log](https://github.com/user-attachments/files/20266449/gklib.5.log)
[moonraker (4).log](https://github.com/user-attachments/files/20266450/moonraker.4.log)
adiing log fyi @jbatonnet

fatalsaint, 2025-05-19:
This is being tracked here: https://github.com/jbatonnet/Rinkhals.apps/issues/13 - however, the printer just isn't powerful enough at this time to do the timelapses.  It'll shutdown with an MCU timer error.  It's possible the ffmpeg could be re-compiled with hardware acceleration to help but that hasn't been done yet (to my knowledge).

rowcount, 2025-05-20:
I have a problem only with detection gcode from this module, Timelapse initiated and works, but klipper mismatch gcode macro for him.