jbatonnet/Rinkhals #188: USB printing -> ignores start gcode

tom-321, 2025-05-20:
I dont know if this is possible by rinkhals but I have major wifi issues and try to workaround with printing over USB.

So far so good -> now if I hit print on the display it ignores my custom startcode 

[image]

jbatonnet, 2025-05-21:
Hi, thank you for the feedback. Unfortunately, all the USB priting is done by Anycubic and there's not really anything that con be done in Rinkhals to help here.
If you want to make sure, just switch to stock and try again but I'm pretty sure you'll have the same issue with the stock firmware :/

Something you can do is to use Mainsail to upload your files maybe. Support for files larger thatn 100MB is coming in next release.

closed 2025-05-21 by jbatonnet

tom-321, 2025-05-22:
Hi, I'm currently not understanding this completely: is there running klipper on this printer or not ?
And why we can't just use klipper screen instead of this unhandy anycubic UI ?
Are there some restrictions made by anycubic or is it just for using GoKlipper so we can participate by the software updates ?

Is it generally possible to use stock klipper on the hardware of the S1 ?

tom-321, 2025-05-22:
> Something you can do is to use Mainsail to upload your files maybe.
Yeah I've just managed to get the wifi working. So thats now the workflow 👍

>  Support for files larger thatn 100MB is coming in next release.

That's great :)