jbatonnet/Rinkhals #190: Installer worked, but after reboot the screen didn't accept touches

dleffler, 2025-05-20:
**Using Installer on K1S worked, but afterward...**
I used the new installed to install Rinkhals 20250517_01 on my K1S where I installed/upgraded from Rinkhals 20250506_01 under firmware 2.5.2.3.

**To Reproduce**
Steps to reproduce the behavior:
1. Download the installer package
2. Place on the USB drive in the correct folder and rename to update.swu
3. Follow the on-screen prompts to check for updated version, manage, then update
4. Updated version will download then automatically reboot the printer
5. After reboot, the screen appears normal, but wouldn't accept any touch input anywhere on the screen
6. I rebooted/recycled the printer and then everything worked as expected

**Logs / Additional information**
None, it was a one-time event?

aikiharp, 2025-05-21:
I'm having the same issue, but my screen doesn't register touches even after a reboot.

jbatonnet, 2025-05-22:
The fix is coming in the next release

jbatonnet, 2025-05-23:
Fixed in last release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250523_01