jbatonnet/Rinkhals #201: Crash at print start. Kobra 3, Rinkhals 20250526_01_test, firmware 2.4.0.4.

IguanaBob, 2025-05-27:
**Describe the bug**
On my Kobra 3 I just updated to Firmware K3_2.4.0.4 and then Rinkhals 20250526_01_test. Now, when I start a print right after the prep finishes and at the start of the actual print job the printer just stops and the web UI gets a disconnect error.

Screen doesn't show an error, but trying to cancel does nothing.

[image]

Error in Web UI:

[image]

Print head is stuck at the purge area:

[image]

I am able to reboot from SSH but not from the web ui.

**To Reproduce**
Steps to reproduce the behavior:
1. Start a print from the touch screen. Leveling can be either on or off, it happens either way.
2. Observe it get stuck after prep step complete.

**Logs / Additional information**
I am still able to get a debug bundle with USB. I am attaching the bundle:

[debug-bundle_2295568684_20250527-161821.zip](https://github.com/user-attachments/files/20462995/debug-bundle_2295568684_20250527-161821.zip)

root@Rockchip:/useremain/rinkhals/.current/logs# tail app-moonraker.log 
[kobra.py:request()] - [Kobra] Injected objects list
[kobra.py:patch_status()] - [Kobra] Converted Kobra state standby
[kobra.py:_request_standard()] - [Kobra] Injected Mainsail macros
[kobra.py:patch_status()] - [Kobra] Converted Kobra state heating
[kobra.py:patch_status()] - [Kobra] Converted Kobra state printing
[job_state.py:_status_update()] - Job Started: pencil holder Kobra3 v3.gcode
[kobra.py:patch_status()] - [Kobra] Converted Kobra state printing
[kobra.py:patch_status()] - [Kobra] Converted Kobra state printing
[job_state.py:_status_update()] - Job Started: pencil holder Kobra3 v3.gcode
[klippy_connection.py:_on_connection_closed()] - Klippy Connection Removed

gklib.log:

panic: interface conversion: interface {} is int, not int64 [recovered]
	panic: interface conversion: interface {} is *runtime.TypeAssertionError, not string

goroutine 14 [running]:
k3c/common/utils/sys.CatchPanic()
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/common/utils/sys/SysUtils.go:134 +0x374
panic({0x61e4d0, 0xd5bea8})
	/opt/go/src/runtime/panic.go:785 +0xfc
k3c/project.(*cs1237).cs1237_calibration_DataProcess_handle(0xc7a2c8, 0xd24d20)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/cs1237.go:308 +0x4e4
k3c/project.(*SerialReader).taskThread(0xc94e00)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:202 +0x6fc
k3c/project.(*SerialReader).bg_thread(...)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:119
created by k3c/project.(*SerialReader)._start_session in goroutine 1
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:253 +0xfc
panic: interface conversion: interface {} is int, not int64 [recovered]
	panic: interface conversion: interface {} is *runtime.TypeAssertionError, not string

goroutine 14 [running]:
k3c/common/utils/sys.CatchPanic()
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/common/utils/sys/SysUtils.go:134 +0x374
panic({0x61e4d0, 0xe166f0})
	/opt/go/src/runtime/panic.go:785 +0xfc
k3c/project.(*cs1237).cs1237_calibration_DataProcess_handle(0xc7a2c8, 0xccd5c0)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/cs1237.go:308 +0x4e4
k3c/project.(*SerialReader).taskThread(0xc94e00)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:202 +0x6fc
k3c/project.(*SerialReader).bg_thread(...)
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:119
created by k3c/project.(*SerialReader)._start_session in goroutine 1
	/data/liuxiaobo/k3_sys_cartesian_test/1.0/k3c/project/serialhdl.go:253 +0xfc

Darkcats77, 2025-05-27:
the new firmware most likely changed the paths to the log files or others... after resetting to factory settings and removing the rinhals, everything works

IguanaBob, 2025-05-28:
"Did you try reapplying 2.4.0.4 after the downgrade?"

No but I'll try later when I have time.

IguanaBob, 2025-05-30:
Since the downgrade back to 2.4.0 I've tried 2.4.0.4 again and Rinkhals 20250529_01_test, both via update.swu method. Same symptoms. Here's a new debug bundle.

[debug-bundle_2295568684_20250530-050039.zip](https://github.com/user-attachments/files/20515250/debug-bundle_2295568684_20250530-050039.zip)

Thanks.

nolantern, 2025-05-30:
Hey I upgraded to Kobra 3 firmware 2.4.0.4 and Rinkhals 20250529_01_test today. Additionally I added the changes of the Anycubic Next Slicer (1.3.5.2) to my Orca Slicer (2.3.0) and now I'm printing a benchy successfully.

I print with the ACS and currently it's equipped with one spool.

## Changes in Orca Slicer
#### Printer Settings/Machine start G-code
added following lines at the end:
```
; print_bed_min = {print_bed_min[0]},{print_bed_min[1]}
; print_bed_max = {print_bed_max[0]},{print_bed_max[1]}
; print_bed_size = {print_bed_size[0]},{print_bed_size[1]}
; first_layer_print_min = {first_layer_print_min[0]},{first_layer_print_min[1]}
; first_layer_print_max = {first_layer_print_max[0]},{first_layer_print_max[1]}
; first_layer_print_size = {first_layer_print_size[0]},{first_layer_print_size[1]}
```
#### Printer Settings/Change filament G-code
Maybe not related but added following lines:
```
; FLUSH_START
;;; M400 P0
T[next_extruder] ; change extruder
; 1
;;; G1 E-2 F2400
;;; G1 Z{max(toolchange_z+0.6, 5.0)} F600
;;; G1 X271.5 F12000
;;; G1 E-5 F600
;;; G1 X278 F600
;;; M400 P3530
;;; G1 E-1 F600
;;; G1 Z{max(toolchange_z+2.6, 7.0)}  F600
;;; G1 X278 F18000
;;; G1 X271.5 F1200
;;; G1 E2 F300
;;; M400 P31050
;;; M400 P17510
; 2
;;; G1 X278 F18000
;;; G1 X271.5 F1200
;;; G1 E2 F300
;;; M400 P0
;;; G1 X250 F1200
;;; M400 P0
;;; G1 X278 F18000
;;; G1 X271.5 F1200
;;; G1 E2 F300
;;; M400 P0
; 3.1
;;; M106 S255
;;; G1 X36 Y260 Z-0.8 F9600
;;; G1 X50 Y260 Z1 F9600
;;; G1 X86 Y260 Z1 F9600
;;; G1 X90 Y260 Z-0.8 F9600
;;; G1 X90 Y259 Z-0.8 F9600
;;; G1 X86 Y259 Z1 F9600
;;; G1 X50 Y259 Z1 F9600
;;; G1 X36 Y259 Z-0.8 F9600
; 3.2
;;; G1 X36 Y260 Z-0.8 F9600
;;; G1 X50 Y260 Z1 F9600
;;; G1 X86 Y260 Z1 F9600
;;; G1 X90 Y260 Z-0.8 F9600
;;; G1 X90 Y259 Z-0.8 F9600
;;; G1 X86 Y259 Z1 F9600
;;; G1 X50 Y259 Z1 F9600
;;; G1 X36 Y259 Z-0.8 F9600
; 3.3
;;; G1 X36 Y260 Z-0.8 F9600
;;; G1 X50 Y260 Z1 F9600
;;; G1 X86 Y260 Z1 F9600
;;; G1 X90 Y260 Z-0.8 F9600
;;; G1 X90 Y259 Z-0.8 F9600
;;; G1 X86 Y259 Z1 F9600
;;; G1 X50 Y259 Z1 F9600
;;; G1 X36 Y259 Z-0.8 F9600
;;; G1 Z{toolchange_z}
;;; M400 P0
; FLUSH_END
```
#### Printer Settings/Change filament G-code
Maybe not related changed `M601` to `M600` so it fits the setting in Anycubic Next Slicer v1.3.5.2

## printer.conf
Notice that my `printer.custom.conf` does not contain changed bed mesh settings. Maybe this is also important:
````
# to speed up filament changes
[filament_hub]
default_unwind_speed: 50
default_feed_speed: 50

# to correct spoolman filament usage
[extruder]
rotation_distance :6.68304

# more probes for better z offset and mesh
[probe]
samples: 3

# spoolman config
[gcode_macro SET_SPOOL]
gcode:
    {{ action_call_remote_method("spoolman_set_active_spool", params) }}

[gcode_macro M555]
gcode:
    SET_SPOOL SPOOL_ID={{ params.S }}
````

IguanaBob, 2025-05-31:
Hmm, I finally got back to this issue as my Max showed it on the last print too.... when checking the start g-code I notice it had a line for linear-advance that yours does not have. I wonder if that's related.

G9111 bedTemp=[first_layer_bed_temperature] extruderTemp=[first_layer_temperature[initial_tool]]
M117
M900 K0.051

I'll have to test and compare with and without as well as with the gcode you provided. Thanks. I"ll update with more info when I have it.

IguanaBob, 2025-06-18:
Finally was able to get back to this.

Just updated to K3_2.4.0.4 and Rinkhals 20250617_01. First test print seems to be working, does not crash after prep anymore.

Thanks for everything you do.

Darkcats77, 2025-06-18:
> Finally was able to get back to this.
> 
> Just updated to K3_2.4.0.4 and Rinkhals 20250617_01. First test print seems to be working, does not crash after prep anymore.
> 
> Thanks for everything you do.

Press the button in the menu, full auto-calibration, you will be very surprised later) calibrations will not work and the resonance will be constantly calibrated

IguanaBob, 2025-06-18:
> Press the button in the menu, full auto-calibration, you will be very surprised later) calibrations will not work and the resonance will be constantly calibrated

Maybe, but that was not the subject of this ticket. The issue on this ticket has since been resolved. If you are having another issue I suggest you follow up in the issue related to it instead of unrelated issues.

IguanaBob, 2025-06-18:
Update and reopening:

Updated to K3_2.4.0.4 and Rinkhals 20250617_01. First two prints, both started on the touch screen, one with leveling turned on one with leveling off, both worked without issue. I thought the problem as resolved and closed the ticket.

Thinking I was good to go, I then began working on anther calibration issue. For that, I edited printer.custom.cfg with the following.

[stepper_x]
rotation_distance : 32.2
#rotation_distance :32

[stepper_y]
rotation_distance :30.2
#rotation_distance :32

[stepper_z]
rotation_distance :8.1
#rotation_distance :8

Rebooted, tried another print from the screen, crashed just like before.
I then tried reverting my printer.custom.cfg change and rebooted again, but the problem persisted.

Reverted to 2.4.0 and Rinkhals 20250617_01, test print from screen with leveling enabled: Worked

Tried re-applying printer.custom.cfg changes and rebooted. Test print from screen, with leveling: Worked (and axis-calibration now looks great).

So, clearly there is something still wrong. No sure if it had anything to do with my printer.custom.cfg since reverting the change didn't help, so that may have been a coincidence. Not sure it's related to leveling being on/vs off as I don't always use it (need to test this better). Don't think it's a double-mesh issue and mutables doesn't have more than one mesh. Not sure at all what is going on. I haven't yet tried the full removal + factory reset + reinstall, but I will try this later. For now I'll stick with 2.4.0 and 20250617_01 and try again when I have more time.

Thanks again for all you do.

jbatonnet, 2025-06-18:
> Maybe, but that was not the subject of this ticket. The issue on this ticket has since been resolved. If you are having another issue I suggest you follow up in the issue related to it instead of [...]

Yes @Darkcats77 please keep issues isolated if they are not related. You've made the issue you're referring to clear enough on GitHub and Discord.