jbatonnet/Rinkhals #202: rinhals on the new firmware breaks paths and does not allow normal operation

Darkcats77, 2025-05-27:
all operations related to mutable files do not work, the flow is not calibrated (error IN LOGS) autoflow does not work

Darkcats77, 2025-05-27:
after removing the rinhals and resetting to factory settings, everything works fine!

zerograv90, 2025-05-28:
Please close the ticket if its solved for you. This makes it easier to see which bugs are still unsolved.

Darkcats77, 2025-05-28:
> Пожалуйста, закройте тикет, если он решен для вас. Это облегчает просмотр нерешенных ошибок.

So nothing has been decided! rinhals breaks the 2.4.0.4 firmware, only its complete removal helps! And that's the problem.

Darkcats77, 2025-05-28:
> Извините, тогда это была моя ошибка. Похоже, вы переустановили rinkhals после сброса

I put Rinkhals on a clean, updated 2.4.0.4 kobra 3, and on 2 printers, both started doing resonance checks and stopped adequately calibrating the flow and bed level.. after removing rinhals and resetting the printers to factory settings, everything worked as expected.

Darkcats77, 2025-05-28:
>

> Trying to follow here. What makes you think it's permission related? Can you share the permission list for those files?

I just see that the settings that are in the changeable files are not applied, and they are written there, but it starts printing with a resonance check, as if they are not there, but they are there. There is also an error in the logs regarding the application of flow calibration parameters. But he does the calibration physically. The first layer also comes off, although it obviously calibrates the table, but logically it's all related to the files being modified and the fact that the clipper can't read them, apparently... I'm not claiming, but I deleted the rinhals, then reset the printer settings to factory settings and everything worked fine..

Darkcats77, 2025-05-28:
[gklib.log](https://github.com/user-attachments/files/20489194/gklib.log)

Darkcats77, 2025-05-28:
I put the mod again, for some reason all the settings in the changeable files disappeared, although all calibrations were made on a clean 2404, I put it to print, the resonance did not appear, stopped printing and performed all calibrations through the printer menu, the changeable files were filled with calibration data, I put the print, and immediately a resonance check appeared, although it did not it was enabled, the problem is clearly in rinhals, something is wrong with the paths, why did the old data in the mod files disappear after installing the mod?

jbatonnet, 2025-05-28:
The resonance issue if somehow related to the firmware updates. Not sure why yet... but it can usually be fixed by downgrading and reapplying system updates.
But Rinkhals does not touch the mutable files. That's really weird that they are disappearing.

Can you generate a debug bundle to get a broader context of your printer?

Darkcats77, 2025-05-31:
I deleted everything and installed a new version of the mod, everything is exactly the same, the resonance is constantly being calibrated again, and the mutable files are clean after installation, after calibrations they are filled in, but again I see that it does not work adequately

Darkcats77, 2025-06-01:
> The resonance issue if somehow related to the firmware updates. Not sure why yet... but it can usually be fixed by downgrading and reapplying system updates. But Rinkhals does not touch the mutable [...]

Nothing has changed on the new firmware 2404 and the latest mod either! if you want to repeat the mistake, just run a full calibration from the printer screen, after that it will fill in the mutable file and will not read it before each print, as it should, and for this reason all the data it contains is not applied, and therefore the printer performs a resonance check every time, as well as in the logs you can see that it calibrates the flow but does not apply its data!  as well

UnknownVHHV, 2025-06-01:
I have the same problem.

After installing Rinkhals, when I press "Print" and get to the screen with the options (like leveling, flow calibration, timelapse), the printer gives an error if I turn off leveling. It only works if leveling is turned on.

But with the stock firmware, I could print without using leveling. 

[image]

UnknownVHHV, 2025-06-01:
[debug-bundle_284377474_20250601-084412.zip](https://github.com/user-attachments/files/20538988/debug-bundle_284377474_20250601-084412.zip)

Darkcats77, 2025-06-01:
why does the mutable file become empty after installing rinkhals? despite the fact that it was filled with calibration data in the stock firmware? and the errors only appear after enabling full calibration of the printer and filling in this empty file with new data, I think that's the problem.

avxmw, 2025-06-02:
Same problem here, but I did not yet have time to dig in further.

As for the calibration not being used, I wonder if that has to do with the update changelog stating "2. Support regional leveling" - likely some (variant) of KAMP?

What is the best way to get completely clean leaving as few traces as possible(via SSH) before a fresh setup of Anycubic firmware?

Darkcats77, 2025-06-02:
> Та же проблема, но у меня пока не было времени вникать глубже.
> 
> Что касается неиспользуемой калибровки, интересно, связано ли это с тем, что в журнале изменений указано «2. Поддержка регионального выравнивания» — вероятно, это какой-то (вариант) KAMP?
> 
> Каков наилучший способ полностью очистить систему, оставив как можно меньше следов (через SSH) перед новой настройкой прошивки Anycubic?

remove the mod via the installer from the USB flash drive. Then, after a reboot, clean the system by resetting to factory settings through the printer menu, and everything will be fine and clean.

jmmerida, 2025-06-09:
The resonance testing issue also happens in 2.4.0 with the 0523 release

Darkcats77, 2025-06-09:
> The resonance testing issue also happens in 2.4.0 with the 0523 release

there is not only a resonance problem, the mod simply does not work with firmware 2404 and others where there is an adaptive grid.

Darkcats77, 2025-06-16:
It was found that the problem with enabling resonance measurement is related to enabling FULL calibration from the printer menu with one click, if you calibrate one at a time, the problem does not occur! The full calibration function was added to the firmware quite recently, can you look towards the rinxcals configuration related to this item? Turn on full calibration and you will get the error that I am writing to you about above, this applies to everyone who seems to have rinhals working without problems.

Darkcats77, 2025-06-18:
> I have the kobra3 and kobra 3 max On kobra3, the resonance is still giving me trouble. The k3m is fine, just mechanical issues.

There is not only resonance, there are no automatic calibrations, no table, no flow!

joelpr1234, 2025-06-22:
> if you calibrate one at a time, the problem does not occur! 

How? I have this resonance test each time I print and I did each test separately but the resonance test keeps showing up. :(