jbatonnet/Rinkhals #207: Webcam Error in Fluidd/Maisails

maframan, 2025-06-01:
**Describe the bug**
A clear and concise description of what the bug is, and what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to home page of fluidd or mainsails
2. See Webcam error 

The printer lan mode is disabled.
Is it possible to have the webcam stream active both in the app and in Fluidd/Mainsail?

Darkcats77, 2025-06-01:
> Можно ли сделать так, чтобы трансляция с веб-камеры была активна как в приложении, так и в Fluidd/Mainsail?
> 
> **Чтобы воспроизвести** действия по воспроизведению поведения:
> 
> 1. Перейти на домашнюю страницу fluidd или гротов
> 2. См. Ошибка веб-камеры
> 
> Режим локальной сети принтера отключен.

explore the klipper wiki

jbatonnet, 2025-06-01:
If by app you mean the Anycubic apps then no it's not possible. You can check the FAQ about that: https://jbatonnet.github.io/Rinkhals/faq/

closed 2025-06-01 by jbatonnet