jbatonnet/Rinkhals #212: Remote Display outside its network?

quinman24, 2025-06-08:
I'm trying to use your remote display outside of my home network like on a model phone network. is there a way to control my kobra 3's screen from my phone from outside my home/network?

jbatonnet, 2025-06-08:
Hi, there are a couple of ways to achieve that:

1. Using OctoEverywhere. There's a Rinkhals app you can install from this repo: https://github.com/jbatonnet/Rinkhals.apps
It allows you to connect to your printer remotely. There's a free tier. OctoEverywhere is compatible with RInkhals but the project is not related.

2. Use a VPN from the printer. You can setup a Tailscale or Cloudeflare VPN and there are compatible apps for Rinkhals. You can install VPN client for those services directly on the printer and use them remotely. You'll need some VPN setup knowledge for that.

3. Use an external VPN. If you have a external setup in your home, on a server or another machine you have. Then you can just connect to your VPN and access your printer using it's local IP. Again, you'll need networking knowledge.

NetSplitter is a small networking tool, but it won't be able to help you for that.

quinman24, 2025-06-08:
no the new remote display feature you know the VNC server that you recently added that lets you see your printer screen remotely not mainsail or fluidd.

closed 2025-06-10 by jbatonnet