jbatonnet/Rinkhals #213: Kobra 3 V2 Compatibility?

jonathanalevi, 2025-06-09:
Hi! I'm wondering if you know about Kobra 3 V2 compatibility, and if not, would you like me to test it, as I think I'm probably one of the first folks to have one?

mircsicz, 2025-06-09:
just saw your Video about it yesterday and immediatly hoped you'ld try ;-)

jbatonnet, 2025-06-11:
Well I was able to validate that Anycubic kept the same update structure and password.
Here is a test release with K3V2 support: https://github.com/jbatonnet/Rinkhals/releases/tag/20250611_01_test

I'll wait to get the first feedbacks to release it.

jbatonnet, 2025-06-17:
Available in latest release: https://github.com/jbatonnet/Rinkhals/releases/tag/20250617_01

closed 2025-06-17 by jbatonnet

alienboyxp, 2025-07-21:
Hi @jbatonnet just tested on K3 v2 and seems not working... if you need anything please let me know.