jbatonnet/Rinkhals #219: Dynamically expose webcams and resolutions

jbatonnet, 2025-06-13:
- Tweak app.json on startup to expose
- React to new/old webcams

doctorevil30564, 2025-06-18:
Is it possible to add something in for mainsail / fluid so we can stop and start the mjepg streamer app for when /if we need to run a print through the workbench in the anycubic slicer? I know it's possible in regular klipper, but I know that mjpeg streamer is not running as a service, so it would need to be done as a gcode or something as a kludge to do it.

doctorevil30564, 2025-06-18:
or alternately, is there a way to expose and use the stock camera setup as a url in the mainsail / fluidd web interface?

jbatonnet, 2025-06-29:
Done in https://github.com/jbatonnet/Rinkhals/releases/tag/20250629_01