jbatonnet/Rinkhals #224: 5x5 mesh to 10x10 for auto leveling

joelpr1234, 2025-06-15:
This is not an issue, but I would like to share that for the K3, I use a 10x10 mesh instead 5x5. The probing is better and makes better first layers. I assume this can work in other models, just change two settings and you're good. Just place this code below in the printer.custom.cfg and save it, reboot, then run the auto leveling from the LCD menu and done.
 
New code:

[bed_mesh]
speed: 120
fade_start: 1
fade_end: 10
fade_target: 0
horizontal_move_z:4
mesh_min: 15,15
mesh_max: 240,235
probe_count: 10,10
mesh_pps: 3,3
algorithm: bicubic
split_delta_z: 0.012

Original code:

[bed_mesh]
speed: 120
fade_start: 1
fade_end: 10
fade_target: 0
horizontal_move_z:4
mesh_min: 15,15
mesh_max: 240,235
probe_count: 5,5
mesh_pps: 2,2
algorithm: lagrange
split_delta_z: 0.012

[image]

jbatonnet, 2025-06-18:
I'll close this one for now and as mentioned I welcome any contribution in this regards.
Thanks for sharing, I'm sure many people on the Discord would appreciate having a tested configuration to reuse.

closed 2025-06-18 by jbatonnet

joelpr1234, 2025-06-18:
> There's a lot of discussion in the discord channel for rinhals. Maybe you want to join there (if you havn't yet)? There are also some recommendations around. Some are even playing with the sensivity for the sensor. But maybe this is something for the documentation? Just to show the possibilitys. Like configurable enhancements/opened doors which are only possible with rinkhals?

I don't have discord. I will have to open an account... thanks

joelpr1234, 2025-06-18:
> I'll close this one for now and as mentioned I welcome any contribution in this regards. Thanks for sharing, I'm sure many people on the Discord would appreciate having a tested configuration to reuse.

no problem. thanks.