jbatonnet/Rinkhals #227: 2.4.0 Firmware: Kobra 3 resonance calibration and power failure

Gdenich, 2025-06-16:
Dear Rinkhals friends, I need your advice.
From the end of last year I had 2.3.5.3 firmware and Rinkhals builds up to February on my Kobra 3. Then I had less time for 3d-printer tuning and firmware testing, I just used it for printing, a lot of stuff.
This was a good working machine, but then I decided to test the latest Rinkhals firmware, as a lot of progress was made, and I found that the latest builds work with latest stock firmwares. As 2.4.0.4 has still problems with Rinkhals, I updated to 2.4.0.
Soon my good working machine appeared to be the non-operating machine:

- **10129 Y-axis vibration compensation abnormal:** I had never had it before, I faced it the first time after 2.4.0 installation, during first resonance calibration procedure, it completed only from the second run. Then I found out, that every print from Orca started to run this resonance calibration procedure before every print, so I had only a couple of successful prints, and then, on other prints I had only "10129" error and stop of print process.
The main problem is that I could not disable this resonance calibration procedure before every print, so I could not print at all! 
I then removed Rinkhals firmware with the installer on the USB-drive, and made the factory reset of the stock firmware. The resonance calibration procedure still gives the error now (0129 Y-axis vibration compensation abnormal), but at least it stopped to do it before every print, and I can make the prints actually.
I have checked the wiring of the Y-axis accelerometer, reattached the connector. Checked the weld contacts on the accelerometer board across the wire terminal on the motherboard, each wire is testing good. I will try to do the reattaching of the connector on the motherboard, but actually it is under antivibration sealant compound there, I do not expect much from this operation;
- **power interruption detected**: the printer rebooted and offered to resume printing. This is the thing I had never seen before, even at very long prints. This is not the PSU error, for sure. 

I have these problems all in one day, so I think stock firmware update has something to do with all this. 
Does anyone have any negative experience with the new firmwares? Is it feasible to downgrade the stock firmware, and check, how it works, and is it possible to make a clean downgrade?

MJFairbairn, 2025-06-19:
Install 2.4.0  with Rinkhals 20250523_01 and that should work ok.
I had exactly the same problems with 2.4.0.4

1. Uninstall Rinkhals
2. Install 2.4.0
3. Restore clean Firmware (factory reset)
4. reboot m/c
5. Do Setup
6. Install Rinkhals 20250523_01

That should work.

JoshTheDerf, 2025-07-21:
Does this mean the issue is fixed now? This issue has been driving me nuts for weeks.