jbatonnet/Rinkhals #230: After system reset (factory reset) with installed Rinkhals default leveling can't find bed mesh

Felixoid, 2025-06-19:
**Describe the bug**
Hello. I've had a lot of experiments, firmware updates, and so on. After resetting the system with the installed firmware 2.4.0.0 and Rinkhals [20250531_01](https://github.com/jbatonnet/Rinkhals/releases/tag/20250531_01), it's impossible to print anything from the printer display without checked `Auto leveling`, it gives the error [11532](https://wiki.anycubic.com/en/error-codes/11532-code) ` Device lacks automatic leveling`.

I tried every type of leveling, but nothing can solve the issue:

- Leveling from the printer settings (3d icon on the display)
- Button `CALIBRATE` from the Moonraker's `Heightmap` screen
- Macros `BED MESH CALIBRATE` from the `Dashboard` screen (I think it's the same as the previous)

**To Reproduce**
I am not sure if it's not an edge case, but so far:

1. Have installed Rinkhals 20250531_01 and firmware 2.4.0.0
2. Reset the printer
3. Finish the initial setup
4. Print a model without selected `Auto leveling`

I will try another reset, if you need.

**Logs / Additional information**
Here are klipper [gklib.log](https://github.com/user-attachments/files/20824604/gklib.log) and moonraker [moonraker.log](https://github.com/user-attachments/files/20824734/moonraker.log)

The first has errors related the mesh:

```
2025-06-19T17:27:03.239Z DEBUG project/extras_bed_mesh.go:1901 Bed Mesh state has been saved to profile [default]
for the current session.  The SAVE_CONFIG command will
update the printer config file and restart the printer.
2025-06-19T17:27:03.240Z DEBUG project/Configfile.go:965 start save config
2025-06-19T17:27:03.241Z WARN project/Configfile.go:637 Can't read autosave from config file - modifications after header
2025-06-19T17:27:03.308Z WARN project/Configfile.go:637 Can't read autosave from config file - modifications after header
...
2025-06-19T19:02:34.766Z ERROR project/extras_virtual_sdcard.go:850 printer_mutable.cfg is no exist
```

Felixoid, 2025-06-19:
I connected to processes of `gklib`, `gkapi`, and `K3SysUi` with `strace`, and at the moment the probe has finished, the K3SysUi shows the access to the following file:

`open("/userdata/app/gk/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)`

The following trick did the magic:

```
root@Rockchip:/userdata/app/gk# ln -s printer_data/config/printer_mutable.cfg .
root@Rockchip:/userdata/app/gk#
```

It feels like Moonrake and K3SysUi use different configurations. Before the reset, I had the wrong mesh applied during printing the models. It looked like the following  complete mess:

[image]

Now with printing from the moonrake, I had a completely different result.

Felixoid, 2025-06-20:
Today, I launched the strace more broadly. 

```
> adb shell strace -e file -ffp 713 | grep /printer_mutable.cfg # process ./gklib -a /tmp/unix_uds1 /userdata/app/gk/printer_data/config/printer.generated.cfg
[pid   727] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1336, ...}, 0) = 0
[pid   727] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 20
[pid   727] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 21
[pid   727] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config//printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1336, ...}, AT_SYMLINK_NOFOLLOW) = 0
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg.temp", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE|O_CLOEXEC, 0666) = 20
[pid   727] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1336, ...}, 0) = 0
[pid   727] renameat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable_20250620104330.cfg") = 0
[pid   727] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", 0xf4b4d0, AT_SYMLINK_NOFOLLOW) = -1 ENOENT (No such file or directory)
[pid   727] renameat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg.temp", AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg") = 0
[pid   727] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1342, ...}, 0) = 0
[pid   727] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 20
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 21
[pid   720] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1342, ...}, 0) = 0
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 20
[pid  3624] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 21
[pid  3624] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config//printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1342, ...}, AT_SYMLINK_NOFOLLOW) = 0
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg.temp", O_WRONLY|O_CREAT|O_TRUNC|O_LARGEFILE|O_CLOEXEC, 0666) = 20
[pid  3624] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1342, ...}, 0) = 0
[pid  3624] renameat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable_20250620104457.cfg") = 0
[pid  3624] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", 0xe1e170, AT_SYMLINK_NOFOLLOW) = -1 ENOENT (No such file or directory)
[pid  3624] renameat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg.temp", AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg") = 0
[pid   720] fstatat64(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", {st_mode=S_IFREG|0644, st_size=1337, ...}, 0) = 0
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 20
[pid   720] openat(AT_FDCWD, "/userdata/app/gk/printer_data/config/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 21
```

```
> adb shell strace -e file -ffp 733 | grep /printer_mutable.cfg # process ./K3SysUi
[pid   733] open("/userdata/app/gk/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 15
[pid   733] open("/userdata/app/gk/printer_mutable.cfg", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 15
```

That means gklib and K3SysUi use different configs.

Then, I've read the K3SysUi strings:

```
root@Rockchip:/userdata/app/gk# strings K3SysUi | grep printer_mutable
printer_mutable.cfg
cp -fd /userdata/app/gk/printer_mutable.cfg /userdata/AcLog
cp -fd /userdata/app/gk/printer_mutable_* /userdata/AcLog
printer_mutable.cfg
rm -rf /userdata/app/gk/printer_mutable_*
cp -fd /userdata/app/gk/printer_mutable.cfg /userdata/AcLog
cp -fd /userdata/app/gk/printer_mutable_* /userdata/AcLog
printer_mutable.cfg
printer_mutable.cfg
printer_mutable.cfg
rm -rf /userdata/app/gk/printer_mutable*
printer_mutable.cfg
rm -rf /userdata/app/gk/printer_mutable_*
```

Adding a symlink is the least evil.

I also wanted to report it to the support so that the bug can be resolved in the firmware. But from the following line in `files/3-rinkhals/start.sh`, it looks like the Rinkhals changes the config to launch gklib:

`files/3-rinkhals/start.sh:nice -n -20 ./gklib -a /tmp/unix_uds1 /userdata/app/gk/printer_data/config/printer.generated.cfg >> $RINKHALS_ROOT/logs/gklib.log 2>&1 &`

Creating the symlink is one of the forgotten steps to apply during the installation, right?

jbatonnet, 2025-07-21:
Hi @Felixoid, thank you for the feedback. K3SysUi is the process responsible for the UI, and TBH I never assumed it was reading the printer mutable file as well. The path seems to be hardcoded in the binary.
I'll try to use a bind mount instead of a symlink, or a symlink if it doesn't work

JoshTheDerf, 2025-07-21:
This might explain the issues I've been having with horrible/no auto-leveling, and nozzle crashing into the bed.

jbatonnet, 2025-07-28:
Available in https://github.com/jbatonnet/Rinkhals/releases/tag/20250725_01

closed 2025-07-28 by jbatonnet