jbatonnet/Rinkhals #246: Kobra S1 MJPEG Webcam not working

Olli0073, 2025-06-27:
App MJPEG start und ended immedialty on Kobra S1.

No

riotdd, 2025-06-29:
Running the nightly buld as well with an older Firmware 2.5.3.1 and it's not working neither.

jbatonnet, 2025-06-29:
Hi, more information is needed such as logs, ideally a debug bungle to diagnose.
I've applied the bug issue template to your issue.

Olli0073, 2025-06-29:
Hi.

Reinstalled Rinkhals. Then installed latest Test Version. Webcam ist now working.

jbatonnet, 2025-06-29:
Hi, perfect then, thank you for the feedback!

closed 2025-06-29 by jbatonnet

riotdd, 2025-06-29:
I'm seeing this in the logs.

Module libcamera unavailble, import failed
2025-06-27 00:39:05,905 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method _Connector.on_timeout of <tornado.tcpclient._Connector object at 0xa4f69770>>)
Traceback (most recent call last):
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
          ^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 173, in on_timeout
    self.try_connect(iter(self.secondary_addrs))
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 127, in try_connect
    stream, future = self.connect(af, addr)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 316, in _create_stream
    socket_obj = socket.socket(af)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 232, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol
2025-06-27 01:09:05,832 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method _Connector.on_timeout of <tornado.tcpclient._Connector object at 0xa4d2e190>>)
Traceback (most recent call last):
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
          ^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 173, in on_timeout
    self.try_connect(iter(self.secondary_addrs))
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 127, in try_connect
    stream, future = self.connect(af, addr)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 316, in _create_stream
    socket_obj = socket.socket(af)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 232, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol
2025-06-27 03:39:06,386 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method _Connector.on_timeout of <tornado.tcpclient._Connector object at 0xa4f49a90>>)
Traceback (most recent call last):
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
          ^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 173, in on_timeout
    self.try_connect(iter(self.secondary_addrs))
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 127, in try_connect
    stream, future = self.connect(af, addr)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 316, in _create_stream
    socket_obj = socket.socket(af)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 232, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol
2025-06-27 07:39:06,417 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method _Connector.on_timeout of <tornado.tcpclient._Connector object at 0xa4dac4d0>>)
Traceback (most recent call last):
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/ioloop.py", line 750, in _run_callback
    ret = callback()
          ^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 173, in on_timeout
    self.try_connect(iter(self.secondary_addrs))
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 127, in try_connect
    stream, future = self.connect(af, addr)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 316, in _create_stream
    socket_obj = socket.socket(af)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 232, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol
2025-06-27 09:09:05,887 [ioloop.py:_run_callback()] - Exception in callback functools.partial(<bound method _Connector.on_timeout of <tornado.tcpclient._Connector object at 0xa4dacf10>>)
Traceback (most recent call last):
[... cut ...]
    stream, future = self.connect(af, addr)
                     ^^^^^^^^^^^^^^^^^^^^^^
  File "/useremain/rinkhals/20250625_01_test/home/rinkhals/apps/40-moonraker/lib/python3.11/site-packages/tornado/tcpclient.py", line 316, in _create_stream
    socket_obj = socket.socket(af)
                 ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/socket.py", line 232, in __init__
    _socket.socket.__init__(self, family, type, proto, fileno)
OSError: [Errno 97] Address family not supported by protocol