jbatonnet/Rinkhals #250: Kobra S1 - start ACE Pro Filament Drying feature (Print While Drying) from OrcaSlicer

kingaffiliation, 2025-07-07:
I really appreciate the hard work on Rinkhals. I recently purchased a Kobra S1 combo and Rinkhals allowed me to use OrcaSlicer in Linux.

Apologies if this has already been discussed somewhere - I tried searching and couldn't find this topic.

I'm hoping there's a way to remotely enable the filament dryer function with the ACE Pro. Ideally within OrcaSlicer. Is it possible via gcode? I don't really know how it would work. Is this possible? I just want to be able to enable the Filament Drying feature (Print While Drying) that you see as an option on the Kobra S1 screen when starting a new print locally, alongside "Auto-Leveling", "Flow Calibration", and "Time-Lapse". Thanks in advance!

Anycubic Kobra S1 firmware 2.5.3.8
Anycubic Color Engine Pro firmware 1.3.863
Rinkhals version 20250629_02

jbatonnet, 2025-07-20:
Hi, thank you for your message. The short answer is that GoKlipper cannot be modified to add more features.
But I know someone managed to add arbitrary commands as new macros. I don't know for sure, but we might be able to use that to expose more features as macros someday

kingaffiliation, 2025-07-20:
Thanks for your reply. That sounds good. I've been using Rinkhals web UI to start/stop drying, so any method to get it combined into OrcaSlicer would be amazing. Thanks again for your work on this project.