jbatonnet/Rinkhals #253: file_manager: Error adding inotify watch to root 'gcodes', path: /userdata/app/gk/printer_data/gcodes/3mf_temp

kkeonline, 2025-07-12:
This error occur sometime, but not every time.

Moonraker warnings found.
file_manager: Error adding inotify watch to root 'gcodes', path: /userdata/app/gk/printer_data/gcodes/3mf_temp

[image]

Latest rinkhals 20250706_02
Latest firmware 2.5.3.8
Mainsail: Disabled
The rest: default 
Rinkhals app: Cloud2LanBridge

Printer working fine without problem, only error appear after send print jobs via cloud. (using cloud2lanbridge feature)

phatlemon, 2025-08-16:
Any update or resolution this?

burkminipup, 2025-09-03:
I get this often. I have been on 3 different releases of Rinkhals as well.

kkeonline, 2025-09-03:
I'm leaving rinkhals and revert back to latest stock firmware.
I use rinkhals for stable camera access. but lost some cloud features and timelapse.
Now the latest firmware fix camera access error so I revert back to stock firmware and never got canera access error like the previous firmware.

Djimi, 2025-09-24:
Does someone fix that? I have this too ...