jbatonnet/Rinkhals #260: Tailscale app doesn't work with a Spoolman.

DeusPerpetuum, 2025-07-20:
Hi. I have server setup: headscale, server have tailscale client (for connect local services) and Spoolman server.

My versions:
- printer: Kobra 2 Pro.
- firmware version: 3.1.2.
- Rinkhals version: 20250607_02.

Spoolman config for moonraker have a Spoolman part, server IP is set to my server tailnet ip. Other clients (PC,phone) can connect to the Spoolman, but printer can't. 
Tailscale app from Rinkhals.apps installed, connected to the tailnet (I can see printer on the server and other clients), but moonraker cannot connect to the Spoolman. 
At the same moment Fluid works in my tailnet, I can connect from the other clients.

Also from ssh command: 
```
curl -v http://<:my tailnet server ip:>:<:Spoolman port:>
```
doesn't return anything, like it's doesn't exist. 

I'm really hope you'll help me and others with a same problem.

DeusPerpetuum, 2025-07-20:
SSH command means command execute on the printer through printer ssh

jbatonnet, 2025-07-21:
Rinkahls Tailscale app uses usermode networking as the printer OS doesn't have the right kernel drivers to use standard tunneling.
It means that other devices on your tailscale network can route their packets to the printer, but other apps on the printers cannot route their traffic to the tailscale network.
There's one solution using a HTTP proxy. Based on the documentation, Tailscale can act as a HTTP or SOCKS proxy to route the traffic to the network: https://tailscale.com/kb/1112/userspace-networking#configure-your-application-to-use-socks5-or-http
If you manage to configure Moonraker to use this proxy, you might be able to connect to a "remote" Spoolman then.

martinbogo, 2026-03-30:
BTW -- if I don't hear back from anyone on this thread, I will automatically close this issue w/o action on April 1st.

closed 2026-04-05 by martinbogo