jbatonnet/Rinkhals #271: Please add support for Anycubic Kobra S1 firmware v2.5.6.4

matissrapsa, 2025-08-01:
Anycubic has released a new official firmware update v2.5.6.4 with the following changes:

Fixed the issue of occasional print progress loss

Please consider adding support for Anycubic Kobra S1 Combo firmware v2.5.6.4 in Rinkhals. Thanks.

fenio, 2025-08-04:
You can always downgrade upstream firmware to supported version.

dabassham, 2025-08-04:
> Anycubic has released a new official firmware update v2.5.6.4 with the following changes:
> 
> Fixed the issue of occasional print progress loss
> 
> Please consider adding support for Anycubic Kobra S1 Combo firmware v2.5.6.4 in Rinkhals. Thanks.

Yep, made the mistake of updating today.  Installed on my Kobra 3 Combo just fine, but the S1 doesn't show up on the console menu.

skike-1, 2025-08-05:
> You can always downgrade upstream firmware to supported version.

The new version of cobra s1 does not allow to roll back. Tried several times.

RunRudi, 2025-08-05:
> > The new version of cobra s1 does not allow to roll back. Tried several times.
> 
> I have just rolled back the firmware from 2.5.6.4 to 2.5.6.0 using rinkhals installer. The rinkhals didn't work on 2.5.6.4, it works just fine now.

How did you use Rinkhals for the downgrade of the FW of the printer?

egzumer, 2025-08-05:
> How did you use Rinkhals for the downgrade of the FW of the printer?

From rinkhals installer, install&updates, anycubic, manage and I chose 2.5.6.0. That is it. It downgraded first try.

skike-1, 2025-08-06:
> > > The new version of cobra s1 does not allow to roll back. Tried several times.
> > 
> > 
> > I have just rolled back the firmware from 2.5.6.4 to 2.5.6.0 using rinkhals installer. The rinkhals didn't work on 2.5.6.4, it works just fine now.
> 
> How did you use Rinkhals for the downgrade of the FW of the printer?

I tried again today, now Cobra sees the flash drive simply as a storage device, I formatted it again and still it doesn't see it as an installer but as a simple flash drive -_-

dabassham, 2025-08-06:
Any idea of what features where lost in the firmware rollback? I can't find
what was added in the new AnyCubic firmware

fenio, 2025-08-06:
@dabassham 

[image]

dabassham, 2025-08-06:
Ah, thank you. No big loss.  I think I'll roll back and install.  Much
appreciated.  It installed just fine on my Kobra 3 Combo.