jbatonnet/Rinkhals #272: Mainsail every print status is cancelled

marcel-goldammer, 2025-08-01:
Since version [20250723_01_test](https://github.com/jbatonnet/Rinkhals/releases/tag/20250723_01_test) all of my successful prints get the status `cancelled` in Mainsail. The print however works totally fine.

I am using a AC Kobra S1 printer. Below I uploaded the printer's debug bundle and also a screenshot of my Mainsail print history.

Thank you very much for the work you put into this project!

[debug-bundle_3127837003_20250801-055057.zip](https://github.com/user-attachments/files/21542423/debug-bundle_3127837003_20250801-055057.zip)

[image]

CLNilix, 2025-08-16:
same problem on fluidd. the problem appeared after the last update (Rinkhals 20250725_01 on Anycubic KS1 2.5.6.0).

MichiMess, 2025-08-21:
Yeah, same problem