jbatonnet/Rinkhals #286: When changed probe points 5x5 to 7x7 for bed mesh doesn''t work it will do 5x5 not 7x7.

marcokitt2000, 2025-08-25:
**Describe the bug**
When follow the video from Kanrog Creations i know the printer,generated.cfg file i copy the the mesh and i changed only in printer.custom.cfg file.
[bed_mesh]
#speed: 300
#horizontal_move_z:3
#mesh_min:5,5
#mesh_max:245,245
probe_count7,7 but it does 5x5.
#mesh_pps:2,2
#split_delta_z:0.0125
gorithm:bicubic
add this line extra bicubic_tension 0.3 ? maybe lower or higher
#fade_start: 1
#fade_end: 10
#fade_target: 0

And on the video he setup the camera to full hdd 1080 mode i can only hd ready 720.

**Printer setup**

- Printer firmware version the latest one V2.5.6.4._1.4.0_1.4.0 shows on the screen device
- Firmware version V1,3,863 the combo ace pro
- Rinkhals version latest version it did with 202508-11 or 12 but is now to the latest 20250815_01 on rinkhals app
- anycubic version 2.5.6.4
- maybe other user have the same problem 

I hope this will fix and i hope ther will be a z offset (like normal paper mode setup. or adjust live.

But thanks for this open firmware i like it a lot.

Mvg,
Marco

marcokitt2000, 2025-08-25:
This is crazy i save it and restart firmware it doesn't work. i turn printer off and wait 1 min turn it on now its working crazy.
bit i checked the printer.genereted.cfg now are all things off 7x7 is now bicubic etc and the rest is off to. i will do some checks. but the camera is still 720 not 1080 i think is settings in the mpegstreamer.

jbatonnet, 2025-08-25:
Hi, you can refer to this page for printer configuration issues: https://jbatonnet.github.io/Rinkhals/Rinkhals/printer-configuration/
Unfortunately, config changes are not officially supported within Rinkhals scope as it completely depends on Anycubic implementation.
Now having said that, you can join the Discord server using the invitation listed on the home page, many people are discussing such changes so I'm sure you'll find some leads over there.

closed 2025-08-25 by jbatonnet