jbatonnet/Rinkhals #305: new firmware k3

Zionmatrix2025, 2025-09-28:
I just got updated to 2.4.4.3 on a kobra3 combo
It noted custom firmware which  rarely happens with Rinkhals.
After the update, the tool kit said Rinkhals was disabled, Fix would not work. I tried rebooting after the fix.
I tried the manage screen and it couldn't find a new version.
I tried removing Rinkhals, the grab the latest installer (changed it update.swf), and booted the flash drive and the installer couldn't find anything for Rinkhals.

I love this firmware for monitoring from a web browser and for helping manually level my bed (I have install ender3 spring kit)

I hope there is a new version coming soon!

I have not tried backing down the firmware version, YET.

Zionmatrix2025, 2025-09-28:
Ok, I was using [Rinkhals 20250815_01]
I see there is a test release for 2.4.4.3 - I will try that later this coming week.

Zionmatrix2025, 2025-10-08:
> The Pre-Release 20250919_01_test installer won't start for me on kobra3 combo firmware V2.4.4.3_1.4.0_1.4.0_release.
> 
> I tried both installer and update, chatgpt convinced me to get a USB 2.0 32GB thumbdrive but it didn't help. I get the beep when I plug the drive in but that's it.

I had to do a factory reset before getting to install.

ebspeter, 2025-10-17:
Thanks @RafalBabinicz, fixing the dirty bit didn't change anything for me. I think it's always set when it's been connected to the printer as there's no way of unounting there. 

Curious about how you "also added channel.sh to userdata.img"? 

Though my symtoms are slightly different as I do get the first beep, and I don't get any logs exported to the stick. I can export logs manually but I can't find how to extract the .pack file.