jbatonnet/Rinkhals #306: No pressure advance in test firmware Kobra 3 V2

figgelus, 2025-09-29:
> [!NOTE]
> Please include as much details as possible and help the maintainers. A bug report only saying "X doesn't work" will end up being frustrating for everyone and not helpful.
> Please remember that Rinkhals is a project maintained by a few people giving their free time to the community. Help them help you!

Start print with presssure advance set, if you look in fluid or mainsail pressure advance get set to 0.
Using orcaslicer, going to test print in the morning with nextslicer and see what mainsail etc says for pressure advance.

A clear and concise description of what the bug is, and what you expected to happen.

**Printer setup**

- Kobra 3 V2
- 1.0.9.7
- 2025-09-19_01_test

**Repro steps**

Steps to reproduce the behavior:
1. start print with set pressure advance
2. pressure advance get set from standard to 0 in mainsail and fluiid.

**Debug bundle / Logs**

Generate a debug bundle to collect logs on your printer before posting a new issue.
Instructions are available here: https://jbatonnet.github.io/Rinkhals/guides/generate-debug-bundle/

Satmosftk2, 2025-09-30:
https://github.com/jbatonnet/Rinkhals/issues/296

figgelus, 2025-09-30:
> Same thing https://github.com/jbatonnet/Rinkhals/issues/296 for k2pro

Want to say it's not the same because it changes PA from standard value to zero when prints start. 

I can change value in mainsail when in standby, but not when print in progress.

Satmosftk2, 2025-10-01:
> > Same thing [#296](https://github.com/jbatonnet/Rinkhals/issues/296) for k2pro
> 
> So new test, by changing to marlin based g-code from klipper in orca this problem is gone, can you test to verify [@Satmosftk2](https://github.com/Satmosftk2) If this works for you I want to say it's a problem with the klipper macros to set acc, PA , etc.

KobraOS should work on Klipper based g-code. By using Marlin we lose important functions.
At the moment I'm working in Orca with my post-processing script (python). But accel to decel still not supported from gcode.

```
# Post-processing script for Orca Slicer v2.2.0 and above
# Script version 1.0.2 based on https://github.com/TengerTechnologies/Smoothificator
# Just changes SET_VELOCITY_LIMIT and SET_PRESSURE_ADVANCE on M204 and M900
# For Anycubic Kobra series printers (like Kobra 2 Pro)
import re
import argparse

def process_gcode(input_file):
    # Read the input G-code
    with open(input_file, 'r') as infile:
        lines = infile.readlines()

    # Process the G-code
    modified_lines = []
    i = 0
    while i < len(lines):
        line = lines[i]
		
        # Detect velocity and PA changes
        if line.startswith("SET_VELOCITY_LIMIT ACCEL="):
            velocity_match = re.search(r'([-\d.]+)', line)
            if velocity_match:
                current_velocity = float(velocity_match.group(1))
                velocity_line = re.sub(r'SET_VELOCITY_LIMIT ACCEL=', f'M204 P{current_velocity:.0f} T{current_velocity:.0f} ;Set velocity ', line)
                modified_line = re.sub(r'\w+\s\S+$', f'value for KobraOS', velocity_line)
                i += 1
                modified_lines.append(modified_line)
        elif line.startswith("SET_PRESSURE_ADVANCE ADVANCE="):
            pa_line = re.sub(r'SET_PRESSURE_ADVANCE ADVANCE=', f'M900 K', line)
            modified_line = re.sub(r'; Override pressure advance value', f' ;Override pressure advance value for KobraOS', pa_line)
            i += 1
            modified_lines.append(modified_line)
        else:
            modified_lines.append(line)
            i += 1

    # Overwrite the input file with the modified G-code
    with open(input_file, 'w') as outfile:
        outfile.writelines(modified_lines)
		
# Main execution
if __name__ == "__main__":
    parser = argparse.ArgumentParser(description='Process G-code file to modify velocity and PA for KobraOS')
    parser.add_argument('input_file', help='Input G-code file')
    args = parser.parse_args()
    process_gcode(input_file=args.input_file)
```

Satmosftk2, 2025-10-02:
Just write M205  in Orca start gcode, see more https://github.com/jbatonnet/Rinkhals/issues/296